espruino/BangleApps #979: 2FA Authenticator App icon

Didgeridoohan, 2021-12-01:
@andrewgoz The icon for the 2FA app does not at all work with the dark theme. A 1px white border might be a good idea... Or:

@gfwilliams Maybe there should be an option to have both bright and dark theme compatible app icons (or maybe there already is, but I couldn't see anything about that anywhere). Talking of which, should all objects described on the [theme reference page](https://banglejs.com/reference#l_Graphics_theme) really be the same (3 fb and 3 bg)?

gfwilliams, 2021-12-01:
Ahh, the docs are out of date: https://github.com/espruino/Espruino/blob/master/libs/graphics/jswrap_graphics.c#L3571-L3577

If you make the app icon black and white with a white icon I believe Bangle.js will then use the current theme's FG and BG colours to render it in the right colour

Didgeridoohan, 2021-12-01:
That's right... Now that you mention it I've seen that behaviour when experimenting myself.

andrewgoz, 2021-12-01:
I've just changed it to a colour icon in #981. Looks much better now.

Didgeridoohan, 2021-12-14:
Since the icon was changed, I'm gonna call this one solved...

closed 2021-12-14 by Didgeridoohan