espruino/BangleApps #984: Openstreetmap scaling issue after move to 96x96 tiles

berkenbu, 2021-12-02:
It seems that the map scaling is off after the move to 96x96 tiles in openstmap. The easiest way to see this is by overlaying a recorded, known gps track in gpsrec. Alternatively, it is also evident from the marker being misplaced in the openstmap app itself. 
I briefly looked at the openstmap library code, but nothing obvious stuck out to me. Still trying to understand the ominous factor of '4096' in there (assuming this is related to the tile scale in Openstreetmap)?

gfwilliams, 2021-12-03:
Argh - thanks for the report. I'll have a look at this... As you say nothing seems obviously wrong.

I've got to admit in custom.html I'm not convinced by this line :)

```
scale : 10000*Math.pow(2,16-zoom), // FIXME - this is probably wrong
```

It feels like we should at minimum include the 4096 in `scale`

gfwilliams, 2021-12-03:
Hope that fixed it (in dev app loader now) - let me know!

berkenbu, 2021-12-03:
Gordon, thanks for looking into this so quickly. I just tested the fix, and the scale factor seems accurate now, but there is still an offset that misplaces the marker and/or gps track. I confirmed that the old 64x64 tile version (restored in a private repository) does not have this issue. This is not an urgent issue for me, but I will try and look into it over the weekend.

gfwilliams, 2021-12-06:
Thanks! That is strange... I'd done some debugging on the PC by writing to canvas and I saw and offset there but assumed it was an issue with what I was doing with the canvas. I guess it's possible some offset is needed when writing the tiles into the canvas in the custom.html file

closed 2021-12-06 by gfwilliams

berkenbu, 2021-12-06:
Thanks, looks like it is working correctly now! 
Two possible explanations for the 16px offset: you now use an odd number of tiles (5 by 5 at 96x96 pixels) instead of an even number (8 by 8 at 64x64 pixels), so the center of the map is no longer the corner of a tile, but in the center instead...
Or it could be related to (96-64)/2 = 16 - although admittedly I don't see where that comes into play...
Anyways, thanks for fixing it.

commit 16b92b61 (2021-12-06): openstreetmap 0.11: Add slight offset to OSM data to align it properly
+      Fix alignment of satellite info text (fix #984)