espruino/BangleApps #985: Astronomical step counts on GadgetBridge

aivopaas, 2021-12-02:
For unknown reasons I'm getting quite comical activity reports on GadgetBridge from Bangle 2.
Everything should be on the latest versions.
Haven't tried debugging it yet in any way so I have no idea which component exactly might be causing it (I'm pretty sure it's not my feet though).

[image]

gfwilliams, 2021-12-03:
Wow! Which app are you running on the Bangle? Gadgetbridge or Android?

The step reporting stuff will be in whichever ones of those are used - maybe it's not resetting the counter so is accumulating each time an update is posted

myxor, 2021-12-03:
Glad to see i am not the only one having similar step counts :)
I am using the Active Pedometer and the Android apps on the Bangle.js 1.

CountMurphy, 2021-12-03:
Same here. GadgetBridge and Active Pedometer installed here, though mine only did it once.

CountMurphy, 2021-12-09:
After installing the Android app, my counts are normal/correct

gfwilliams, 2021-12-13:
Fixed with #1058 now!

closed 2021-12-13 by gfwilliams