espruino/BangleApps #991: Locale seems to be broken

T0TProduction, 2021-12-03:
Just got my bangle 2 today, and everything is awesome.
Since I'm from Germany I set the locale to +1 and time showed up correctly.
Now, for no apparent reason, the time that is being displayed ignores locale.
I synced time via the app loader, but no matter what value I choose for locale, the time ignores the offset.
It's not the watchface, the wrong time shows up in the menu when manually adjusting the time as well.

The only thing I can think of that might have changed it was installing, running (unsuccessfully), and then uninstalling the app "GPS time". I'm not sure the issue appeared from exactly then on, but it could have.
Even so, I completely restarted the watch to make sure it's not some runtime issue. Issue persists.

If there's anything I can do to help debugging this, lmk

T0TProduction, 2021-12-03:
So, after resetting the watch ("install default apps"), this was fixed. Even after installing all my currently used apps via "install favourites" it continued to work. 
Can't tell where this went wrong, but seems to me like some app broke something

gfwilliams, 2021-12-06:
That's an odd one - thanks for the report though.

If you can find out how to reproduce this, please let me know. Someone a while back had issues and I wondered whether it was to do with the Web IDE changing the time on upload? Does that ring a bell?

I'm pretty sure the app loader sets everything ok, but it may well be that the IDE doesn't set timezone correctly - I can't remember

elrod16, 2021-12-06:
I didn't have timezone issues with locale but improperly set temperature units. I don't know if this could be connected, as a generalized failure of locale that can happen, or if it is its own bug. I figured it was at least worth mentioning here though.

gfwilliams, 2021-12-06:
Yes - please can you see if `require("Storage").read("locale")` returns anything? That's where the customised locale module should be

elrod16, 2021-12-06:
Unfortunately I already reset my watch, but prior to doing so I did read through the weather app's dependencies and check them and "locale" *WAS* missing.

gfwilliams, 2021-12-06:
Well, Locale doesn't automatically install - there's one built-in which is en_GB, but if you want any other language you need to manually install it.

elrod16, 2021-12-06:
I did do the localization setup but didn't have the weather notification 
app installed initially, so I was unaware of the Fahrenheit functionality. 
At some point though it seems the customized locale got wiped. I don't have 
any data to back this up but I believe during an app install or update that 
locale got set to default.

stephenPspackman, 2021-12-07:
I notice that odd parts of my settings can go away—my theme will reset, my time will flip back to GMT, my selected timezone will be forgotten. I'd originally put that down to the known corruption in my boot sequence (reported elsewhere), but … maybe it's all related in some way.

elrod16, 2021-12-07:
This may or may not be related, but seeing as I think this problem started for me after a flurry of switches between gadget bridge and android as well as notify and notifyfs, I have encountered a similar issue with notify.

After moving to notifyfs and trying to see if gadgetbridge or the Android/Messages combo suited my needs better, when I tried to go back to notify (default) the gadgetbridge installation process installed a nonfunctional notify module.

How I said, could be a totally different issue but I encountered it under the same conditions as my locale failure.

gfwilliams, 2022-02-03:
This is fixed in latest firmwares. There's also a menu option for `utils - > rewrite settings` that will recover from this

closed 2022-02-03 by gfwilliams