espruino/BangleApps #1004: Bangle.js2: iOS messages / integration: how to connect

jeroenpeters1986, 2021-12-04:
Hi, 

on my new Bangle.js2 I have installed the Messages app and the iOS integration app.
I cannot figure out how I use it. How do I connect my phone to that? My iPhone is an iPhone SE with iOS 14. 

I did manage to use the nRF Toolbox to update my firmware to the lastest version 2v10.236, so my device should have the capabilities to connect, but I don't see the bangle showing up on the Bluetooth screen. 

On forehand I would like to say I am sorry to ask such a n00b question, but is there anyone who got this working? I didn't see a readme with this question covered, I managed to search for it.

jeroenpeters1986, 2021-12-06:
I got it working by:
 * downloading webBLE for iphone
 * connecting it with the bangle
 * after this first initialisation the iphone appearantly got acquinted with the bangle JS
 * while connected to webBLE it asked me if i wanted to show the notifications on the watch, and this was wahat I needed

closed 2021-12-06 by jeroenpeters1986

gfwilliams, 2021-12-06:
Glad you sorted this out! Could you not just pair it from your iPhone's bluetooth menu after the iOS app had been installed?

jeroenpeters1986, 2021-12-06:
Hi @gfwilliams, no, it didn't appear in the Bluetooth devices list for a while. I am not sure what happened. But after the connection with webBLE, it did :-D