espruino/BangleApps #994: UTC time with firmware 2v10.236

Saduff, 2021-12-04:
After updating to the latest firmware 2v10.236, setting time from the App Loader or when connecting Gadgetbridge sets the time to UTC time. However, looking under Settings -> Locale, the time zone is correct. Tried rebooting the Bangle, but the issue persists.

T0TProduction, 2021-12-04:
Pretty sure this is a duplicate of https://github.com/espruino/BangleApps/issues/991

Saduff, 2021-12-04:
The issue is similar. However, everything was fine for me on firmware 2v10.219 mentioned in #991. The problem started after the firmware update.

T0TProduction, 2021-12-04:
If the issue appears even on earlier firmware, it seems to me it's not related to your firmware switch at all though

Concerning the use of GPS time, as mentioned in that ticket, I am not sure if that was actually the source of the issue. 

Did you test wiping all apps? As it fixed the issue for me, it would be interesting to know if it does so for you as well

Saduff, 2021-12-05:
"Install default apps" did fix the issue for me too. And another display issue which is a bit hard to describe. Both issues started right after the firmware update though, so there seems to be a causal link in my case at least, but maybe the time issue can be caused by something else too.

gfwilliams, 2021-12-06:
In your case, could the issue be that Gadgetbridge is setting the time in the wrong timezone? I can imagine that could be a possibility?

Saduff, 2021-12-06:
No, I tried disabling time sync from Gadgetbridge and used the button in the About tab of the App Loader, but same issue.

I think the issue was that some configuration was corrupted or something. Apart from the time issue, I noticed two other odd things:
<br>
1. The display issue I mentioned earlier. I saw `> bluetooth` and `> -bluetooth` at the bottom left of the screen whenever bluetooth was connected/disconnected. Thought it was a new feature of the latest firmware. Later thought it was debug info, but debug info was set to hidden from settings. Tried setting it on and off too but no change. This wasn't the display issue I was referring to though. I guess that could be described by horizontal lines everywhere on the screen. More showed up over time. Holding down the button for 3 seconds to reload cleared them up.
2. LCD timeout was set to 10 seconds in settings, but it timed out after 5 seconds.

All of these issues were fixed by resetting to default/favorite apps.

gfwilliams, 2021-12-07:
Ok, thanks - I think you're likely hitting this issue then: https://github.com/espruino/Espruino/issues/2109

I'll try and get a fix in for it soon, but until then if you connect with the Web IDE and type load("bootupdate.js") in the left-hand pane (then long-press the button to reload the clock) I bet it'll be fixed?

gfwilliams, 2021-12-07:
Ok, I've just hopefully fixed  #2109 so if you update to the latest Bangle.js firmware now, hopefully this won't occur any more.

Saduff, 2021-12-07:
Okay, I upgraded to 239 firmware and everything is fine after the upgrade.  :slightly_smiling:

I'll close this issue then. Thanks, Gordon!

closed 2021-12-07 by Saduff