espruino/BangleApps #999: Messages app: Notifications won't vibrate?

holycow234234, 2021-12-04:
Not sure if this is a setting I have to turn on or what?
Quiet Mode is off.
Vibration is on.
Vibrate in messages settings is set to --- and 4s.
The motor works as it vibrates when going though menus, etc.
I'm messages 0.07

elrod16, 2021-12-04:
I got it working by uninstalling gadgetbridge watch app but keeping notifyfs

holycow234234, 2021-12-04:
That didn't work for me. 
Something else of potential interest is after uninstalling Android+Messages, and reinstalling GadgetBridge, vibration wouldn't work for notifications until I rebooted the watch

CountMurphy, 2021-12-05:
Same issue here. Moving back to gadgetbridge app until fixed.

gfwilliams, 2021-12-06:
Are we saying that this is likely some issue caused by having *both* Gadgetbridge and Android installed at the same time?

closed 2021-12-06 by gfwilliams

holycow234234, 2021-12-06:
For me at least, it was one or the other.
Anyway, your fix worked for me. Thanks!

CountMurphy, 2021-12-06:
The fix worked for a little bit, but now it no longer buzzes. Buzzer is working for other apps. I've completely uninstalled android and messages and then reinstalled them. No buzz on email (gmail) or duo pushes.

gfwilliams, 2021-12-06:
Are you using the development app loader, because the fix I did this morning isn't yet live on banglejs.com/apps?.

commit 1c9e9725 (2021-12-06): messages 0.08: Fix rendering of long messages (fix #969)
      buzz on new message (fix #999)