espruino/BangleApps #1012: Weather clock shows just black screen

myxor, 2021-12-06:
The new weather clock app just shows a black screen on my Bangle.js 1.

I have installed the weather app and it shows the current weather conditions reported by Gadgedbridge.

I will ping the creator of the clock: @jg76379

jg76379, 2021-12-06:
Good point, I'll try to test in the emulator later.

jg76379, 2021-12-06:
I tested in the Bangle.js 1 emulator and it is working (although it could use some formatting tweaks).

[image]

closed 2021-12-06 by gfwilliams

jg76379, 2021-12-06:
@gfwilliams looks good to me! Thanks for the optimization. It still renders the same on my Bangle 2, and looks okay on the 1 in the emulator.

I thought it might be a memory issue since the Bangle 1 has much less memory. Interesting that that extra wrapping vertical layout caused the issue, i'll watch out for that in the future.

myxor, 2021-12-06:
Can confirm that it works on Bangle.js 1 now. Thanks!

jg76379, 2021-12-06:
@myxor let me know if you have have any more issues or suggestions for the app!