espruino/BangleApps #1027: [gbmusic] Allow play/pause to work without music info

Saduff, 2021-12-08:
When using the Gadgetbridge Music Controls app with Spotify, no music info is displayed (e.g. track, artist) and also the state (playing/paused) is unknown. Because of that, the app is always sending the play command. I propose to toggle play/pause on every touch (alternating between the two) when music info cannot be obtained. All other controls (previous/next track, volume up/down) work fine.

gfwilliams, 2021-12-08:
What device is this on? Because usually I'd say just use Android/Messages and there's a specific entry for music playing which displays all the info

Saduff, 2021-12-08:
The device is Xiaomi Mi 11. But I thought Android/Messages is only for notifications or am I missing something? And just realized that music info also comes from a notification. Is Messages supposed to show it? In any case, I'm more interested in the music controls than the music info, but of course it's better to also have the music info.

Also, I realized why music info didn't show up with Spotify. I had Gadgetbridge configured to only allow notifications from selected apps and Spotify was not selected. After adding Spotify to the list, the music info showed up and the music controls app is working properly. However, it might still be worthwhile to implement what I suggested in case some unknown Android music app does not show a notification about what's playing. At least all the controls should still work then.

gfwilliams, 2021-12-09:
> I thought Android/Messages is only for notifications or am I missing something?

The idea is the `Android` app handles all Android integration (including music) - there are music controls too. It's basically a replacement for Gadgetbridge, but the idea is rather than duplicating a bunch of work for Android and iOS, the two apps (Android/iOS) forward the info to the Messages app, and then we can have one app to handle music/messages/etc that doesn't care what phone you have.

But yes, good point on the toggling. It could toggle automatically and if Gadgetbridge *did* update play state, it would just overwrite it with the same.

Saduff, 2021-12-09:
Oh, I see. I found the music controls in the Messages app now. However, I think it's missing some features that the Gadgetbridge Music Controls app has:

1. Volume control
2. Previous song (I can only see a button for next song)
3. Auto start when playing music

gfwilliams, 2021-12-10:
Ahh - yes, good point. It'd be nice to add things like that. I guess prev/next/vol could be done with swipes?

rigrig, 2022-02-21:
I don't have an iPhone, but it looks as if using `Bangle.musicControl` should work universally.   
If so I think we should probably get rid of the `gbmusic` app, and just move that code into `messages`.

That also allows us to get rid of the "override `GB` for music events" stuff, and handle receiving a "real" message while playing music smoother. (i.e. go back to the music screen after reading the message)

gfwilliams, 2022-02-21:
Definitely - I'm all in favour of trying to reduce duplicated functionality

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
If gbmusic is kind of deprecated by now, I suggest closing this. Maybe keep it and change same or open a new issue regarding adding 'previous'-action and swipes for input - but some of these things are already on the way in the different new message ui projects, e.g. @rigrigs's smessages.

closed 2022-12-02 by Saduff