espruino/BangleApps #1028: Anton clock: truncated text with German locale

neshanjo, 2021-12-08:
**This is a good first timers' issue!**

Given: Bangle 2 with Anton app. German language is installed with the language app.

When: Anton clock is opened.

Expected: Second line shows date (e.g. "8. Dezember")

Actual: Second line shows weekday and date (e.g. "MITTWOCH, 08. DEZEMBER") which, however is truncated, so only "**CH, 08. DEZEMBE**" is visible.

Note that the third line already shows the week day.

I'm probably able to fix this myself. However, I wanted to open this issue such that others can already have a look, since my spare time is currently very limited.

gfwilliams, 2021-12-08:
Thanks - actually the issue seems to be that the German locale is 'wrong':

https://github.com/espruino/BangleApps/blob/master/apps/locale/locales.js#L133

Pretty much every other language uses the shortened month name and NOT day of week, but the German one tries to do everything on one line.

Not really sure what we do here - I guess we make the German date a bit more concise

neshanjo, 2021-12-08:
+1 for using `%d. %b %Y` as format.

I counted 28 other usages of locale.date(), but I guess, it is still fine to change since

- not so many users have the German locale set
- a then shorter string usually makes no problems
- one cannot rely on the length of the date (long version) anyway.

neshanjo, 2021-12-08:
Is there a possibility to set the locale in the emulator? This way, I could quickly check some of the app that use locale.date without having to install them all on my device.

gfwilliams, 2021-12-08:
No, I'm afraid it's not easy to emulate the locale right now :( It would be possible in the future if we changed the way the emulator was called (embedding it in the app loader rather than the IDE) but right now it's not really possible.

neshanjo, 2021-12-08:
Well, I tried ~10 different clocks, including Floral Clock and Wave Clock which also look much better with the shorter date string. Pull request will follow...

gfwilliams, 2021-12-08:
Fixed - thanks!

closed 2021-12-08 by gfwilliams