espruino/BangleApps #1032: [Messages - Feature Request] - make the 3 buttons at the bottom of the view message larger

hughbarney, 2021-12-08:
When viewing a message there are 3 buttons at the bottom of the screen.  These are a little small and a bit hard to accurately tap.  They could do with being 1/3 larger in widdth and height.

gfwilliams, 2021-12-08:
Yes - I think this could be handled just by changing the Layout library actually. I guess buttons that are supposed to be pressed should have a minimum size

nebbishhacker, 2021-12-11:
Does `fillx` work on buttons? It would be nice to have them evenly subdivide the available space.

gfwilliams, 2021-12-13:
It should do, yes... I guess in some cases it might be nice not to do that but I think it could just be added for the messages app at minimum.

gfwilliams, 2022-02-03:
I think icons are actually bigger now? Could you confirm? 
[image]

hughbarney, 2022-02-04:
Apologies - I tend to use my googlemail account for github notifications and only read it every now and again.
Will give this a test soon.

hughbarney, 2022-02-19:
Have finally tested.  Yes these are better.

However I personally dont see much being added by these button at all and would rather have the additional display space on the screen.

<- goes back.   I'm perfectly happy with clicking the back button, which seems intuitive to me.
Tick - haven't a clue what this actually does.
X - haven't a clue what this does or why it is needed.

The act of displaying a message on the screen is all that is needed to send any acknowledgement back to gadget bridge.  Nothing else is needed in my opinion and the Messages app would be simpler / more intuitive without these buttons.

But the original bug can be closed.

closed 2022-02-19 by hughbarney

nebbishhacker, 2022-02-19:
> Tick - haven't a clue what this actually does.
> X - haven't a clue what this does or why it is needed.

"✓" is equivalent to tapping on the notification on the phone, so e.g. if the notification is a text it'll open the messaging app on your phone.
"X" dismisses the notification on your phone.

I personally use the "X" and "<" buttons regularly, depending on whether I want to dismiss a notification completely or deal with it later.

gfwilliams, 2022-02-21:
Link to the forum post you started too: http://forum.espruino.com/conversations/373174/#comment16408087

What the buttons do is in the README for the messages app