espruino/BangleApps #1038: [Android Integration] Find phone not working

Saduff, 2021-12-08:
As mentioned in the title, the "find phone" function in the Android app on the Bangle 2 is not working, the phone does not do anything. The phone is a Xiaomi Mi 11 on MIUI 12.5 (Android 11).

I found [this](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Find-phone) Gadgetbridge page where they say:

> In order for the Find phone feature to work correctly on Android 10 and up, Android requires that the pairing between the mobile device (phone) and the smart bracelet/watch is done via Companion Device Management, read about it here for more details.

From what I understand, that just means pairing the watch with the phone so it shows up in the phone's Bluetooth settings as paired. However, on [this](https://www.espruino.com/Gadgetbridge) page, it's recommended not to pair. Anyway, I tried pairing it, but find phone still didn't work after that.

As a side note, the other way around (to find the watch) works.

gfwilliams, 2021-12-09:
Thanks for the update!

> Update: I found that it works only when Gadgetbridge is in the foreground on the phone. 

Could this be related to battery saving optimisations? I know Gadgetbridge does say you need to disable battery optimisation or the phone will try and shut it down to save power

elrod16, 2021-12-09:
Being that it does do the notification acknowledging the "find my phone" request but stays silent, could it be a "Do Not Disturb" setting issue or the notification is set low priority? I know Google has been rolling out "Adaptive Notifications" on newer Android builds. It adjusts the priority without user input, possibly downgrading Gadget bridge.

Saduff, 2021-12-10:
Thanks for the suggestions. I did enable Autostart and disabled battery optimizations for Gadgetbridge. It seems the issue is not that it gets killed in the background since the notification does appear. I don't see any notification priority options in MIUI. I have my phone on silent all the time, but not "Do Not Disturb" mode. If I turn silent off, the notification makes a bubble sound like all other notifications, but Gadgetbridge still doesn't ring.

Anyway, I'll close this issue soon since this is a Gadgetbridge/MIUI issue and maybe open an issue with Gadgetbridge.

closed 2021-12-17 by Saduff