espruino/BangleApps #1040: Flip Screen/Left handed mode/Button on Left

ruswerner, 2021-12-09:
I've searched the wiki, the docs, GitHub and all the settings on the Bangle 2 and can't figure out how to flip the screen so I can wear the watch on my right arm (for left-handed users). The button should be on the left when looking at the watch.

Is this something that was considered? If not, can you please point me in the direction of where I would have to modify the firmware to accomplish this?

Cheers

gfwilliams, 2021-12-09:
There's nothing in there right now, but you can implement this basically by going to the `custom boot code` app and entering:

```
(function() {
  g.setRotation(2);
})();
```

However, quite a few other things (touchscreen, launcher, etc) don't work too well, so I think this would require some actual firmware changes to do nicely - inverting touch coordinates/directions and also fixing `g.setClipRect` when rotated

thyttan, 2022-04-10:
Too me this could make sense for us wearing the watch on left wrist as well. As it is now you can accidentally press the button when you angle your hand towards the watch. Having the button on the other side of the screen would eliminate that risk.

On a traditional watch it makes sense to have the wheel on right side for right-handed people since it's easier to turn it that way. But with a button there isn't really any benefits to it as i see it, other than familiarity. 

Another note, if implemented, would this mess up accelerometer readings?

gfwilliams, 2022-05-03:
Just saw another request for this at http://forum.espruino.com/conversations/375520/#comment16505901 so there's definitely interest.

I just made a change to the main Espruino firmware at https://github.com/espruino/Espruino/commit/f772bd51062b84626c595c395538182409bb77ff so that now the touchscreen obeys `g.setRotation(2);` - so actually if you stick that in, the watch is pretty much usable left-handed now.

However there's still an issue with scrollable lists - I'll look into that. But once done we can make it an option in `settings`

> Another note: If implemented, would this mess with accelerometer readings?

Right now they are left as-is. I guess it depends what you intended to use the readings for. Step counting/etc would work fine.

gfwilliams, 2022-05-03:
Ok, text rendering issue is now fine - so if you do as mentioned in this comment: https://github.com/espruino/BangleApps/issues/1040#issuecomment-989662278

The Bangle should now work pretty well left-handed.

thyttan, 2022-10-12:
Just testing what you suggest in https://github.com/espruino/BangleApps/issues/1040#issuecomment-989662278 on FW 2v15.31.

Touch screen-touches seems to work pretty good in apps, but behaves weirdly on the widget field.

Targeting the widgets is a little hard and touching might fire them twice with one touch. 

When trying to press the red back button it will (more often) fire the "Light switch widget" which sits on the other side of the widget field - the other way around does not happen. It is possible to hit the red back button as well though.

When rotation=2 and using "Light switch widget" to change brightness level it works for the most part. But if draging to the very top of the touch screen the brightness will be set to minimum value instead of maximum value - it seems some touch-pixels at the very bottom hasn't followed along in the rotation to become top-pixels, if that makes sense.

Swipe/drag directions seem to not have updated. So e.g. scrolling menus, or navigating Desktop Launcher becomes inverted. However as said above, drag works mostly as expected for draging on "Light switch widget" to change brightness.

But just reiterating my edit [from before](https://github.com/espruino/BangleApps/issues/1040#issuecomment-1094394024): 
>Edit: The bangle 2 button isn't as pronounced as the bip one. I have yet to press the bangle button on accident.

So at least for me this is not important if you have other things to prioritize.

thyttan, 2022-10-20:
Just tested the suggestion in https://github.com/espruino/BangleApps/issues/1040#issuecomment-989662278 again (Fw 2v15.47

Regarding https://github.com/espruino/BangleApps/issues/2181#issuecomment-1285236524, 

>Swipe/drag directions seem to not have updated. So e.g. scrolling menus, or navigating Desktop Launcher becomes inverted. 

still stands. But other than that everything else works much better - as good as with normal rotation as far as I can tell.

thyttan, 2022-10-21:
Sorry, drag events works as expected - only swipe events have note rotated along with the screen on rotation==2.

(I had 'Swipe menus' installed when I tested yesterday so menus acted on swipes instead of drags..)

gfwilliams, 2022-10-21:
Ok, fixed now! I guess we're ok to add the setting.

It might also be worth allowing 90 degree rotations to be used too, since it can be nice if the Bangle is on a charging dock.

In those cases, there still seems to be some issue with clipping when scrolling, but it's still usable for most things

thyttan, 2022-10-21:
Just,

>EDIT: I have[ a branch where left handed setting is introduced under LCD settings](https://github.com/thyttan/BangleApps/tree/rotate-setting/apps/setting). **But I don't know where the code for rotating the screen should go** - in setting.js, in firmware or somewhere else?

So I don't know how to do it completely :P I can try to work it out eventually though, but not right now.

gfwilliams, 2022-10-21:
Just dump it in here: https://github.com/espruino/BangleApps/blob/master/apps/boot/bootupdate.js#L94

```
if (s.leftHanded) boot+=`g.setRotation(2);\n`;
```

It might be nice to store the rotation amount though, so it can go sideways.

Also in your settings I'd change `g.setRotation()` on onchange, do a clear, and force a redraw of the widgets. The issue is that if you change to left-handed and change back, we wouldn't get a `g.setRotation(0);` otherwise (and we don't really want it in boot code all the time).

gfwilliams, 2022-10-21:
Actually I'm doing this now...

gfwilliams, 2022-10-21:
Just sorted - it's in the development app loader: d49728fa63946544d23b4c461778adf0cc081678

You can rotate all different ways, and can even mirror it (might be cool if you're looking at it through a mirror for some reason!)

closed 2022-10-21 by gfwilliams