espruino/BangleApps #1046: Pebble bug

RomanistHere, 2021-12-09:
https://github.com/espruino/BangleApps/tree/master/apps/pebble - not sure where to submit things like that.

### Description

White bar at the top of the screen.

### Steps to reproduce

Accidentally happening both at the app and when set as default clock.

### Thoughts

1. Probably other app interfering
2. Just a bug

### How to resolve

If it's because of another app, probably should "margin-top", so this top bar will always be visible

gfwilliams, 2021-12-09:
Thanks! It's great having some shots of the issue. Hopefully fixed now!

closed 2021-12-09 by gfwilliams

RomanistHere, 2021-12-09:
It's not fixed - I just installed it. 

P.S. Got my watch today, very cool :)

gfwilliams, 2021-12-09:
Did you use the development app loader? Because banglejs.com/apps doesn't update immediately

RomanistHere, 2021-12-09:
Not familiar yet, will check dev version and update the status here. Thanks for the responding :)

commit 52ed2c3b (2021-12-09): 0.04: Fix widget hiding code (fix #1046)