espruino/BangleApps #1049: Weather conditions not matching an icon on non-english languages

myxor, 2021-12-10:
When using Gadgetbridge and the Weather notification app on android (as suggested on https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Weather) to push weather conditions to the Bangle.js the weather object 'weather.json') looks like this in German:

```
{ "hide": true, "expiry": 14400000,
  "weather": { "temp": 276, "hum": 74,
    "txt": "Ein paar Wolken",
    "wind": 7, "wdir": 113,
    "loc": " -- ",
    "time": 1639131366761.5966796875,
    "wrose": "se"
   }
 }
```
You can see that the textual representation of the condition is in German: "Ein paar Wolken" ~ "Some clouds".

The weather app does string comparison with the english words of weather conditions to find the matching icon:
https://github.com/espruino/BangleApps/blob/master/apps/weather/lib.js#L207

```
if (condition.includes("thunderstorm")) return drawThunderstorm;
if (condition.includes("freezing")||condition.includes("snow")||
  condition.includes("sleet")) {
  return drawSnow;
}
[...]
```

Any idea how we can get the condition matching working in multiple languages?

myxor, 2021-12-10:
Probably this is a problem which should be fixed upstream in the weather app. There is already an issue in the repository but development seems to have stopped: https://github.com/gelin/weather-notification/issues/56

nebbishhacker, 2021-12-11:
Matching on the weather text was always a bit of a hack - ideally GadgetBridge would include some sort of language-independent condition code or icon name in the weather object.

gelin/weather-notification#56 appears to be more about being able to control the language returned by OpenWeatherMap, which I guess would allow you to avoid the issue by always setting the language to english.

See also #541

myxor, 2021-12-13:
I created an issue on the Gadgetbridge git: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2504

myxor, 2021-12-19:
Looks like we will get the weather conditions code in the next release of GadgetBridge: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2511

thanks @rigrig

myxor, 2022-01-01:
Recently released version 0.64.0 contains change
* Bangle.js: send weather condition code to device

I will have a look at updating weather app code later. If someone else wants to do it feel free.

Didgeridoohan, 2022-01-05:
Haven't looked into the details yet, but updating to Weather 0.14 (and using Gadgetbridge 0.64.0) completely removed weather condition icon on my watch. Only the temperature is shown...

Will look closer as soon as possible, but wanted to give a heads-up.

Didgeridoohan, 2022-01-05:
Might have something to do with Gadgetbridge sending the weather code 804 when that doesn't seem to be available in the weather app (or should the `default: return drawBrokenClouds` deal with that situation? It doesn't though)...

And I also see that there's a fallback to using the old txt method of the code isn't available, but that also seems to be broken because I was on Gadgetbridge 0.63.1 when updating to Weather 0.14 and the icon was broken then as well.

myxor, 2022-01-05:
@Didgeridoohan sorry to read that it does not work on your side. Sadly i am not able to reproduce the behaviour which you are getting. I debugged it on my watch. I tried condition code 804 and saw that it stepped into the `default: return drawBrokenClouds` case and showed the broken clouds icon.
Also when i run it without the condition code it run into the "old" fallback function and drew the icon from there.

Can you debug it on your side?

Didgeridoohan, 2022-01-05:
I've been doing some very light debugging by trying to do a fully clean install of the Weather app, but nothing so far. And I see that it probably doesn't have to do with the reported code (at least not directly) since a code 600 also fails.

I'll try to examine things better at a later time.

rigrig, 2022-01-05:
Only the widget was broken, not the app: the library was expecting an object now, but the widget still used `w.txt`.
#1214 fixes it.

closed 2022-01-05 by gfwilliams