espruino/BangleApps #1053: Flow app does not work on bangle1.

Ronin0000, 2021-12-11:
When I launch the flow app on my  bangle.js 1 there is just a bunch of colored lines going across the screen.

gfwilliams, 2021-12-13:
I picture would be handy, but it is possible it's to do with your watch firmware. There was an issue on 2v10 I think where 3 bit images weren't rendered right (they were all stripey) - cutting edge (or 2v11 when released) will be better

Ronin0000, 2021-12-17:
Here is a picture and I have 2v11 firmware. Sorry for the bad quality picture.
[image]
.

TheGLander, 2021-12-30:
I experienced it on the Bangle.js 1 web emulator, but it worked on my genuine Bangle.js, which I believe wasn't on the latest version (I think some version of 2v11). I think it's an artifact of `Bangle.setLCDMode("120x120")` which I need to use for collision checking. I am pretty sure it's a thing I can't fix, sorry. Probably something firmware related, as Gordon said.

thyttan, 2022-11-28:
Candidate for being closed - leave a comment if you object!

Rationale:
Old issue which I suspect probably have been resolved by one of the subsequent FW updates.

gfwilliams, 2022-11-30:
Just had a quick try installing Flow on 2v14 and it seems the problem still exists, so I guess this needs looking into more

closed 2024-05-02 by gfwilliams