espruino/BangleApps #1069: FW Update App 0.02 not working

hughbarney, 2021-12-12:
Followed instructions on the README.
Had 2.10.241 installed.
After attempting to reflash to 2.10.244 the FW reported was still 2.10.241

jg76379, 2021-12-12:
Yeah I tried to use it to install 2.10.244 and mine is still saying 2v10.219.

gfwilliams, 2021-12-13:
Thanks for letting me know. Do you think you'd be able to get a video of what happens as the watch reboots? So the firmware upload should get to 100% and then it'll reboot to the bootloader - and I'm interested to see what it says

johan-m-o, 2021-12-13:
@gfwilliams Here you go: https://youtu.be/nuD1dRdn9m8

gfwilliams, 2021-12-13:
Perfect - thanks! I'll look into this with the 2v19 bootloader and see what's wrong

gfwilliams, 2021-12-13:
Quick update on this: it's actually down to hardware differences! It seems maybe most of the KickStarter Bangles use a slightly different flash chip which takes a while to initialise. The one on the device I was testing with was able to initialise immediately, but it seems on other watches the bootloader starts, goes to access memory and then finds nothing there because the chip hasn't initialised yet.

So, I'm going to have to provide a bootloader update to fix this - for now, until this is super well tested, I'm going to re-tag the firmware updater as beta and stop it being put right at the top

Didgeridoohan, 2021-12-13:
@gfwilliams Just give a holler if you need help with further testing.

johan-m-o, 2021-12-14:
Well, I just now tested updating through the App loader from 2v10.247 to 2v10.248, but no go... Was expecting this commit to have fixed things, but maybe more is needed?
https://github.com/espruino/Espruino/commit/0a9f07a0fc21d1f989adaa8f35f74b29349307d6

gfwilliams, 2021-12-15:
This is expected - it's why I'm leaving this issue open, it's not really fixed yet.

>  Is that as expected or is something not updating as it should?

You have three parts to Bangle.js: The softdevice (which we're not changing), bangle.js/espruino, and the bootloader.

If you compiled everything from scratch and put it on a device via SWD right now, it'd work great. The problem is all the devices out there still have the old bootloader which has the issue.

So the old bootloader needs updating before firmware updates will work. This can be done by the firmware update but it's really important that everything works reliably since if you break the bootloader you're really screwed. And that's why I'm not rushing pushing out a fix :)

danielo515, 2021-12-15:
Just got my watch and tried to install this watchface... just to be super confused by not being able to activate it, selecting it over and over again on the config with no luck.
Maybe you can remove or flag it on the app loader page?

gfwilliams, 2021-12-15:
@danielo515 sorry, not sure I understand. You were expecting that the firmware updater is a watch face?

danielo515, 2021-12-15:
sorry, I don't know how I ended here. I clicked on the github link of the watch face I was having problems with, saw this issue and I thought it was the watchface not working with the latest firmware. How should I report a problem with one specific watchface?

johan-m-o, 2022-01-11:
@gfwilliams I've been testing FW updater 0.03 and it's been working great. Really nice way to update the firmware...

Had an issue when updating to 2v11.14 though. The update stopped halfway because the watch lost the connection to the App loader. I went back in and restarted the update, which finished apparently without issue. But, when the watch rebooted it got stuck in a bootloop, getting as far as "Booting..." and then rebooting. Loading the watch without any code by holding BTN1 didn't help, but reflashing the same firmware version through nRF Toolbox got the watch up and running again.

Some kind of safety for when connection is dropped midway through the update might be needed.

gfwilliams, 2022-01-11:
Hi - thanks - that is very strange though. The firmware has a CRC in it, so if the written firmware doesn't match the CRC (eg if it's only half-written) then the watch shouldn't update itself. Not to mention that each time you update, the storage file should be erased first and written from scratch.

I wonder whether the app hex file was corrupted somehow when it was downloaded from the website? Even then it's in a zip which should have some CRCs of its own though.

johan-m-o, 2022-01-11:
I can give it a go at replicating, if you'd like. Is there any way I could pull any interesting info off the watch as the update happens?

gfwilliams, 2022-01-11:
If you enter `Puck.debug` in the dev console when using the BangleApps loader, that can help to show what's being uploaded and what the responses from the Bangle are... but yes, if you can get it to happen again, please let me know!

I guess it's possible that while the CRC checked out, there was a flash read error during flashing and that meant that the wrong data was written to the Bangle...

johan-m-o, 2022-01-12:
@gfwilliams The fun times continue... I just attempted to update to 2v11.21 using the Firmware Updater and it failed and factory reset my watch in the process.

For some reason I couldn't connect to the watch on my laptop to use the dev console when updating (it's very old and have had bluetooth issues before), so I used my phone instead (which I usually do). After initialising the update there was a quick error message in the App loader (which I unfortunately did not managed to see what it said, something about a faulty address I think). The upload just stopped and upon reloading the watch it said "No clock". I rebooted and was greeted by the Welcome app and a factory reset watch.

I did have a logcat running as I was updating (since there's no dev console in the Android version of my browser, Vivaldi) but that doesn't seem to have saved for whatever reason.

I'll continue to test this and will report back...

Saduff, 2022-02-03:
Tried the Firmware Update in the App Loader for the first time seeing it wasn't marked as beta anymore, but it failed. Here's what happened:

Prior to update:
* FW: `2v11`
* Bootloader: `2v11.52`

Tried to update to `2v12`. Everything seemed fine, the watch rebooted after upload was complete, went into DFU mode or something similar. Happened pretty fast, so couldn't read everything. But then when it booted up, it was still on `2v11` like nothing happened.

So I decided to try again. This time during the upload of `fwupdate`, I got a "FIFO Full" error or something like that, so I had to long press the button to reload. Tried again and same thing. Finally, on the third attempt, it finished and the watch rebooted. However, now it got into a bootloop. Just saw "Booting..." over and over again.

Then I flashed `2v12` with the traditional method (nRF Toolbox) and that worked and recovered the watch.

gfwilliams, 2022-02-03:
Ok, thanks. That's a bit concerning. Must admit I'm a bit surprised - the bootloader won't start messing with the main firmware unless the CRC of the firmware file matches, so I guess the only thing that could have happened is that the bootloader read invalid data from flash somehow.

Saduff, 2022-02-11:
Could https://github.com/espruino/Espruino/issues/2147 have something to do with the FW update failing? Or maybe causing the "FIFO Full" message when uploading the firmware?

gfwilliams, 2022-02-14:
Ahh - yes, that could be it! The app loader waits a few seconds, but it may well time out if it takes that long to restart the Bangle I guess (although it resets it with `reset()` which doesn't load as much so should be a lot faster than loading the launcher or a clock face)

gfwilliams, 2022-02-14:
Are you finding it works a lot better now you've compacted?

johan-m-o, 2022-02-14:
I've had issues trying to update to both 2v12.27 and 2v12.31. Both times the update stalls at about 25%. Tried compacting and using different browsers (Vivaldi and Chrome), but no luck. Haven't tried resetting my watch, but if it happens again I'll give that a go.

Updating through dfu (nrf toolbox) works fine, but when the watch reboots I see a flashing screen and a message that there's a CRS mismatch and that there'll be no update. Update seemingly went through without issue though...

gfwilliams, 2022-02-14:
> oth times the update stalls at about 25%. 

Is it because you put the app loader in a background tab of the web browser and did something else in the mean time? Web browsers generally stall 'background' webpages after a while.

And yes - you'll get the CRC mismatch because it's complaining about the half-uploaded file the Firmware Update app made - it's not related to the one uploaded by NRF toolbox. Nothing to worry about though.

johan-m-o, 2022-02-14:
@gfwilliams I make sure to leave the browser active and the app loader up front, and also to not let the screen turn off whenever I run the updater (this is on my Android).

I did reboot before running the update the old-fashioned way, right after the stalled attempt, so I expected that the half-downloaded file would have triggered then. But maybe it doesn't trigger on a regular reboot?

johan-m-o, 2022-02-16:
@gfwilliams I'm on bootloader 2v12 and firmware 2v12.31. Attempted to update to 2v12.33, but no go... The update stalls at about 20-25% and just sits there (no errors). Tried doing a full reset of the watch and ran the updater again, with only the default apps installed, but still the same. In other words, it's currently impossible for me to update through the App Loader... Nrf toolbox works without issue.

I'll try grabbing a logcat later, but if you've got some other info you'd like my to grab (from an Android device, I currently do not have access to a computer with Bluetooth) I'd be happy to do that too.

gfwilliams, 2022-02-16:
I don't suppose you have any of the Android debug stuff set up on your PC? If so then you can run chrome on the PC and see the logs that the App Loader is making. If you run `Puck.debug=3` on the console we should be able to see a bunch of info about what's failing and why.

What apps/widgets do you have installed? Could you maybe try uninstalling the Android app temporarily and see if that helps?

Also, if you connected with the Web IDE on your phone and just left it for 10 minutes, it'd be interesting to see if anything gets printed to the console.

johan-m-o, 2022-02-16:
> I don't suppose you have any of the Android debug stuff set up on your PC? If so then you can run chrome on the PC and see the logs that the App Loader is making. If you run `Puck.debug=3` on the [...]

I have previously tried fiddling with remote debugging, but never really got it to work. Will give it another shot.

And I've actually tried with both the default apps and even no apps at all installed and it made no difference.

johan-m-o, 2022-02-16:
Having huge issues debugging this... I have one more data point though: It doesn't seem to be a problem with the watch.

I simply cannot get my Android phone to recognise Chrome when remote debugging from my laptop. USB debugging, ADB and everything else works fine and the device itself is shown under devices when initiating a remote debugging session. But the Chrome instance on the phone does not show up on the laptop no matter what I do... Maybe there's some setting I'm missing somewhere on the phone. My Google-Fu is failing me here though.

So, I figured I would try using my Android tablet instead and it worked flawlessly in getting a debugging session going. But, of course, on the tablet there are zero issues in updating the watch firmware from the App Loader. Went back to the phone and it stalls at the exact same place again.

Excuse me, I'll just be here in a corner pulling my hair out...

gfwilliams, 2022-02-17:
Ok, thanks for looking into this...

> on the tablet there are zero issues in updating the watch firmware from the App Loader

Well, that's helpful actually. So I guess there's nothing installed on your watch that is causing issues (unless the Android table is faster than the phone, so completes in less time and interruptions are less likely?)

What's the Android phone you're having issues with? It's not a custom ROM or something is it?

johan-m-o, 2022-02-17:
> So I guess there's nothing installed on your watch that is causing issues (unless the Android table is faster than the phone, so completes in less time and interruptions are less likely?)

Tablet is actually way slower than the phone... Felt like it took aeons for the update to complete.

> What's the Android phone you're having issues with? It's not a custom ROM or something is it?

It is a custom ROM, Android 11 on a OnePlus 5T. This might be why I'm having troubles getting remote debugging working (I've been searching high and low to find what it could be though), but I've had (almost) zero issues updating the watch firmware from the App Loader until last week when I tried updating to 2v12.27.

I'll keep trying to find a way of getting better info on this...

gfwilliams, 2022-02-17:
Ok - do you get any kind of error message on the watch or displayed in the app? 

About the only thing I can think is the phone maybe sends data *so* fast that the Bangle's flow control doesn't work and some data gets lost, but then you should see a `FIFO_FULL` error message.

I'll see if I can add something to the App Loader so you can just tap and download a debug log from it.

johan-m-o, 2022-02-17:
I don't get any error messages at all on either watch or phone.In contrast to other similar reports one seen on the forum.

I did try setting Debug info on the watch to "log", to maybe catch something there, but nothing. Ever since though, whenever I try updating the firmware the Bluetooth connection is lost whenever the progress bar reaches the point where it previously would stall. Very curious...

Saduff, 2022-04-20:
Tried updating the firmware with the app again, from 2v12 (bootloader also at 2v12) to 2v13.

1. First upload got stuck pretty early (progress bar under "U" of "Uploading")
2. Second upload got stuck with a `FIFO_FULL` error less than halfway through
3. I compacted storage before the third attempt, but the third upload ended up like the first
4. Fourth upload same as first
5. Fifth upload was successful and the firmware update was successful as well

I did the firmware update with an Android phone using Chrome. The phone and watch were right next to each other.

gfwilliams, 2022-04-21:
Ok, thanks - I'd be interested to see if you get any issues when trying the update from another device?

FIFO_FULL happens when more data is received by the watch than can be handled - but there is flow control so it can only really happen if the sending device doesn't use flow control (or is too slow to respond). I guess it might be that the phone you're using is buffering enough data that by the time the XOFF signal is received there's still so much data in the buffer that it overflows.

I've just halved the chunk size - this'll slow down everyone's uploads a little but hopefully will make things more reliable. And I've also just doubled the Bangle.js 2's input buffer size for cutting edge builds (so it'll be in 2v14)

Saduff, 2022-04-21:
> I'd be interested to see if you get any issues when trying the update from another device?

I could try from a Linux laptop. Is it possible to flash the same version or does it have to be newer/different (I think I noticed a diff check during the update)? On that note, is it possible to downgrade? Then I could flash the latest cutting edge build and go back to the release build later if I wanted to.

gfwilliams, 2022-04-21:
You can flash the same version - I think the bootloader itself will just realise and ignore it, but you can easily just test the upload works.

> On that note, is it possible to downgrade? Then I could flash the latest cutting edge build and go back to the release build later if I wanted to.

Yes, totally!

closed 2022-04-21 by gfwilliams

gfwilliams, 2022-04-21:
@johan-m-o is it just Android that you've tried/are having problems with?

Have you ever tried it with a 'factory reset' Bangle? I guess it is possible that something was running in the background on the device (even after the reset) and *that* is causing the instability?

Maybe you could download and attach a copy of your '.boot0' file on the Bangle (or you could do a full backup with the backup tool in the app loader and send that to me so I can see if I can reproduce).

Also, I forgot to mention but the changes I just made are only live on the *development* app loader. So if you tried banglejs.com/apps it'd have still been using the same upload size.

gfwilliams, 2022-04-21:
ahh - but you were also using the new firmware, so that would have made a big difference.