espruino/BangleApps #1082: [Messages/Android Integration] Text messages lead to duplicate notifications

nebbishhacker, 2021-12-13:
When I receive a text message, GadgetBridge sends two separate notifications to the watch, as follows:

```
{"t":"notify","id":1639306871,"body":"Test message","sender":"Ben Whittaker","tel":"+xxxxxxxxxx"}
{"t":"notify","id":1639306872,"src":"Textra","title":"Ben Whittaker","body":"Test message"}
```

I gather the first notification is text-message specific and the second is the generic notification shown by Textra (the SMS app I use).

Both of these notifications are shown in the messages list on the watch, and dismissing the notification on my phone only removes one of the messages on my watch (specifically, the second one with `"src":"Textra"`).

nebbishhacker, 2021-12-13:
Just to verify that this isn't a Textra-specific issue, I tried switching to the stock messaging app, with the same results.

```
{"t":"notify","id":1639306875,"body":"Message body","sender":"Ben Whittaker","tel":"+xxxxxxxxxx"}
{"t":"notify","id":1639306876,"src":"Messages","title":"Ben Whittaker","body":"Message body"}
```

Then when I dismiss the notification:

```
{"t":"notify-","id":1639306876}
```

gfwilliams, 2021-12-14:
That is a strange one... Do you see two distinct notifications on your phone screen?

I guess maybe Gadgetbridge is just trapping an SMS *and* a notification event. Maybe we should just ditch the SMS event?

nebbishhacker, 2021-12-14:
I only see one notification on the phone screen.

If the SMS event is always redundant than ditching it sounds reasonable to me.

elrod16, 2021-12-14:
You need to disable "send SMS" in gadgetbridge settings.

elrod16, 2021-12-14:
I think one of their updates broke some settings database compatibility. Somewhere along the way from v0.60 to v0.63 I noticed all of my notification settings were reset even though other settings were still where I left them.

DDDanny, 2021-12-14:
> Ah, you're right! I must have turned that on at some point and forgotten about it.

Are you sure that you will receive (any) SMS notification anymore now after disabling?
Becasue the setting controls when SMS even's are send (never, at display off, always)

If this has fixed you issue -> no SMS notifications are no more forwarded.
Then this is the scenario that happens:
1. notification: "Missed call" (broadcasted by the telephone app -> gadgetbridge forwards correctly to bangle
2. Do you maybe get a from your mobile provider?`like "Hey you missed Call" SMS -> gadgetbridge will then forward (correctly) to bangle
If you do not see the "missed call" SMS then try to disconnect the phone from cellular network and raise a missed call, because I can imagine that the sms app is clever enough to do some "deduplication" with the phone app notification)

If this is the case you can maybe disable the notifications for missd calls in the telephone app because your provider alert you also when the mobile is not connected to the cellular network.
Or you can maybe disable the "missed call" SMS in the settings of your mobile provider and see if one event disappears.

Or maybe I'm far away from the track ;-)

elrod16, 2021-12-14:
You need to keep Phone on. Only disable SMS. Keep generic notification support on. As long as you haven't blacklisted said messaging apps then their regular old notifications will be forwarded to your phone and be dismissible and not duplicated.

DDDanny, 2021-12-14:
@nebbishhacker - if SMS still works for you, fine
@elrod16 - I recheck and when I set SMS to "never" (german niemals) then I get no more SMS notifications so e.g also the SMS from a 2FA is not notified.... using GB 0.63.0 (via F-driod) - all otheh (non-SMS broadcasts are still working)

I have my personal problem here that active call's are notified to bangle although the phone app is default and phone is set to always - the miss call notifcation works ;) so that's another topic and I hope I can identify the root cause when my compatible bluetooth dongle arrives...

elrod16, 2021-12-14:
The call issue isn't related, there is already a bug report with Gadgetbridge about it and a fix coming down the line on their end.

DDDanny, 2021-12-14:
Do you have an issue id that i can track 👍  ? -> then i can stop search here and focus on the sleep topic

elrod16, 2021-12-14:
It being a Gadgetbridge bug is mentioned here:
https://github.com/espruino/BangleApps/issues/976

You'll have to look at their codeburg page yourself for specifics.

DDDanny, 2021-12-14:
@elrod16 - Thx! 👯 I searched already for "call" on the codeburg and didn't find the corresponding issue...

@nebbishhacker - handover to you if your issue is solved, then this issue can be solved?

nebbishhacker, 2021-12-14:
My issue is solved, yes.

closed 2021-12-14 by nebbishhacker