espruino/BangleApps #1086: LCD backlight not turning on when Bangle2 is twisted

enricoboschi, 2021-12-14:
I'm not entirely sure what the default behaviour should be, but currently the only way I have to turn on the screen's backlight on my Bangle.js 2 is to press the button. Twisting it only seems to unlock the screen (the lock icon in the top left corner disappears), but it remains dark.
Also, is there a way to completely turn off the screen, instead of simply dimming the backlight?
I don't know if it's of any use, but I noticed (using `Bangle.getOptions()`) that changing the setting `LCD TImeout` changes the values `lockTimeout` and `backlightTimeout`, while there doesn't seem to be a way to change `lcdPowerTimeout`.
Firmware: 2v10.219

nebbishhacker, 2021-12-14:
The issue with the LCD backlight not turning on on twist has been fixed in newer firmwares. Follow the [firmware update instructions](http://www.espruino.com/Bangle.js2#firmware-updates) to update to the [latest cutting edge build](https://www.espruino.com/binaries/travis/master/espruino_2v10.246_banglejs2.zip).

enricoboschi, 2021-12-14:
Thanks, that fixed it

closed 2021-12-14 by enricoboschi