espruino/BangleApps #1088: ios app - possible to announce Bangle.js 2 as heart rate monitor to iOS?

ureimers, 2021-12-14:
Hi,
I'm totally new to this (Bangle.js programming, iOS integration) so please excuse that there isn't much technical detail here and that I might say things, you already know.

I was wondering, shouldn't it be possible to announce the Bangle.js 2 as a heart rate monitor source to iOS so that the heart rate red by the health app gets send to (or pulled by) iOS?

The only thing I was able to find, so far, was the keyword `org.bluetooth.service.heart_rate.xml`. It seems Bluetooth defines so called "services" which a Bluetooth device can implement and then announce through an XML, e.g. https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.service.heart_rate.xml?

As I said, I'm new to this and its not easy to even find a starting point here.

Did someone ever try this out? It would be awesome to have the Bangle.js 2 report the heart rate to iOS.

Best regards,
Ulf

ureimers, 2021-12-15:
After I found out about GATT, Bluetooth services, `NRF.setAdvertising` etc., I'm confident, that this can be implemented. So there's my first project :-)

I'm closing this, as I think I answered my question myself.

closed 2021-12-15 by ureimers

gfwilliams, 2021-12-15:
Hi! Thanks! Yes, it should be possible.

It's worth checking out `gpstimeserver` which does basically what you want: https://github.com/espruino/BangleApps/tree/master/apps/gpstimeserver

You just need to copy it, change the UUIDs and push heart rate data in instead.

I'd also strongly suggest using the `NRF Connect` app for debugging as it allows you to really see exactly what services the Bangle is showing and even decodes common data (like heart rate)