espruino/BangleApps #1096: Desktop Launcher - now shows itself in the launcher window ?

hughbarney, 2021-12-15:
screenshot attached
[image]

elrod16, 2021-12-15:
In the app settings there is an option now to filter out launchers and clocks.

hughbarney, 2021-12-15:
Got it.  Didn't realise.  Not a bug.  Will close this.

hughbarney, 2021-12-15:
not a bug, there is a setting to hide them now.

closed 2021-12-15 by hughbarney

elrod16, 2021-12-15:
> Got it. Didn't realise. Not a bug. Will close this.

Yeah I think the reasoning is so that it will show up as launchable in other launchers, allowing toy to test drive it before making it your default. I personally like being able to test out watch faces and launchers without having to set than in case something REALLY doesn't work as intended.