espruino/BangleApps #1099: showimg: icon is blank in the launcher,

hughbarney, 2021-12-15:
Showimg: Thew icon comes up blank in the launcher.  Looks like there is something wrong with the icon format.

It is not clear what this app does.  When you run it you get a blank screen.
There is mention of showimg.user.img but there is no interface to upload a file.
A screenshot would be useful when browsing the App Loader.

elrod16, 2021-12-16:
Yeah the js to version evaluated in the icon isn't right. It doesn't call "heatshrink". As far as the other stuff, you're right about the IDE being the only option. Currently there doesn't seem to be any App Loader interface.

xxDUxx, 2021-12-16:
I just fixed the launcher image, something went wrong on the first version. The app is very simple, no App Loader interface (yet). I will look into it though...

gfwilliams, 2022-02-03:
Icon fixed now

closed 2022-02-03 by gfwilliams