espruino/BangleApps #1105: Messages app: Remove virtual back button

mabuchner, 2021-12-16:
For me the virtual back button in the messages app is really difficult to activate.

Can't we use a short press of the physical button to go back?

Then we wouldn't have this problem and it would give use more space for the message text.

gfwilliams, 2021-12-16:
Press of the button to go back sounds good - however for quite a few messages I think there will also end up being various buttons for actions at the bottom which we wouldn't really be able to do anything about.

There is already an issue open to make the buttons a bit bigger as well, as I know they are a pain to tap at the moment.

DDDanny, 2022-01-15:
as I wrote in #1294 the bangle2 touch functions could seriously enrich ueser experience.
So i suggest double-touch for back-action and a new option to hide back-button by default.
`msg.negative` / `msg.positive` can be handled with swipe left/right and also an option to hide the buttions and instead e.g. make 10 px on the left red and 1ß px on the right green to show to the enduser that  confirm/cancel are availible.

gfwilliams, 2022-02-08:
Just fixed this with 83a849a9d1da0004f8f8141c275efcd25ec4aab2

Physical button now goes back. While we have the on-screen button for now, tapping the text will now bring you to a bigger, scrollable view of the message

closed 2022-02-08 by gfwilliams