espruino/BangleApps #1109: Messages app: Dithered background makes text hard to read

mabuchner, 2021-12-16:
The dithered background in the messages app (with dark theme) sometimes makes it difficult to read the text

[image]

Maybe a solid background colour or at least a solid background behind the text would help.

gfwilliams, 2021-12-16:
Ouch - thanks! I wasn't aware that happened. On light themes the main text's background is plain (like the widget bar) and I'd expected the same here.

mabuchner, 2021-12-16:
I've just opened the same message manually and now the text background uses the same background as the title bar. The message text is then readable, but I'm not sure, if this is the intended behaviour.

gfwilliams, 2021-12-16:
Ok - so how did you open the message first time when you got this to happen?

mabuchner, 2021-12-16:
If I click the bird button and then go back, then the message background is black.

mabuchner, 2021-12-16:
Feels to me like a call to `g.setBgColor` is missing.

gfwilliams, 2022-02-03:
Closing as this has been tweaked now and I feel like it's much improved:

[image]

[image]

closed 2022-02-03 by gfwilliams