espruino/BangleApps #1110: Firmware Updates via Web IDE fails

RomanistHere, 2021-12-16:
I managed to update it via nRF toolbox, however still want to report the issue with web IDE. During ~15-20 tries I got the same two errors:

[image]
[image]

Updating popup didn't go after errors and if console wasn't opened, there'd no chances to understand whether something is going or not. 

Watch showed either "connect, start, error" or "connect, error". Reset to default and factory reset didn't help.

gfwilliams, 2022-02-03:
Thanks - sorry for the delay following this up. Just tried here and it's fine on Linux. I got an error once on Android but then it worked fine - i'll have to see if it's reproducable on Windows.

Did you get the problem every time, or did it work sometimes?

RomanistHere, 2022-02-03:
every time

gfwilliams, 2024-05-02:
Web IDE flasher has now been improved for 6+ months. No reports anywhere of issues now so I think we're good now

closed 2024-05-02 by gfwilliams