espruino/BangleApps #1121: Circleclock icon is 178x178 and takes up whole screen in dtlauncher

hughbarney, 2021-12-17:
Tried to contact the developer through github but can't figure out how to PM on github.
[image]

nebbishhacker, 2021-12-17:
I've noticed a couple other incorrectly sized icons (albeit not that dramatic). I wonder if it'd be worth adding an icon size check to sanitycheck.js.

myxor, 2021-12-17:
Circles clock icon is fixed via https://github.com/espruino/BangleApps/pull/1113

gfwilliams, 2021-12-17:
Thanks! I've been been through using @nebbishhacker's sanitytest additions and have fixed all the massively big or broken icons.

I haven't fixed all the ones that are too big though - I notice you've got a few too @hughbarney!

> Warning: JS icon pebble.img should be 48x48px but is instead 64x64px
> Warning: JS icon rebble.img should be 48x48px but is instead 64x64px

closed 2021-12-17 by gfwilliams

hughbarney, 2021-12-28:
@gfwilliams - oops, its a size setting thats difficult to remember.  It the size you choose works, then you don't question if you got the size right or not.   Maybe a tickbox on the imageInverter that said - [x] this is (48x48) an icon for a Bangle App.