espruino/BangleApps #1125: thermometer 0.05  (bangle1) issue displaying

dapgo, 2021-12-17:
After opening the app i can see 

-  temperature  overlapping top widget area
- sometimes temperature goes to fg and widgets disappear
- other widget are displayed over temperature
- part of the digits of temperature out of  top of the display
- "eading temperature.." string always and placed in the middle of the screen

gfwilliams, 2021-12-17:
Eep. ok - is bootloader up to date on your Bangle 1? I imagine the issue would be if appRect isn't set, but that should be polyfilled by the bootloader

gfwilliams, 2022-01-04:
Ahh - thanks! So I guess the issue is the bootloader file doesn't update when the firmware updates. I'll file a new issue for that.

gfwilliams, 2022-02-03:
This is fixed now - it checks the git commit and reloads if it changed

closed 2022-02-03 by gfwilliams