espruino/BangleApps #1129: Pebbled: - might be a good idea to set battery level warning from a seting as 15% might be too low.

hughbarney, 2021-12-17:
Hi there @RomanistHere.  I see my Pebble watch has been Pebbled :)
Just a note about setting the battery warning to 15%.  

The battery charge tends to fall off a cliff after about 30%.  In the space of 1 hour you can be in 30% the next minute your watch has shutdown as your are out of charge.  I'm wondering if the warning level % should be set from a settings menu ?

RomanistHere, 2021-12-17:
Hi :) 

I haven't experienced it yet. If that's the case I'd suggest another way to inform the user about, like making some parts of screen or elements change colors.

For example, boot and calendar icons become yellow at 30%, and red at 15%.

Or we could move kilometers to the left corner and add battery widget at right one.

hughbarney, 2022-11-05:
dont think this is ever going to get done.  Closing.

closed 2022-11-05 by hughbarney