espruino/BangleApps #1132: Messages: Allow disabling repetition of buzzer

skauertz, 2021-12-17:
I think it should be possible to disable the repetition of the buzzer when a new notification comes in, and set it so it only buzzes once. Currently it can only be set between 2 and 10 seconds.

gfwilliams, 2022-01-04:
Good point - thanks. Yes, definitely makes sense.

skauertz, 2022-01-04:
Yes, that could also be an option. But I would really suggest offering the possibility to just play the buzz pattern once, I think for many users that will be sufficient. After all, the phone itself doesn't continue to buzz either.

jeroenpeters1986, 2022-02-18:
I will come up with a fix for this

jeroenpeters1986, 2022-02-20:
Well.. disabling it isnt hard. The hard part to do this properly is when another app (like menu) is on the screen and we want to have it buzz once, while the repeat is disabled. I tried several (even dirty) pieces of code to provide for this, but im stuck at the moment on this particular situation. So it could take a little bit more time

gfwilliams, 2024-05-02:
Pretty sure you can now just set `repeat` down to 1?

closed 2024-05-02 by gfwilliams