espruino/BangleApps #1138: E.showPrompt fires lcdPower

BartS23, 2021-12-18:
Wenn ich E.showPrompt aufrufe wird das event lcdPower gefeuert. Obwohl das Display an ist.

Ist das ein Bug, oder ein Feature? :)

Input:
```
Bangle.on('lcdPower', on => console.log("lcdPower: " + on))
E.showPrompt("Test")
```

Output:
```
=Promise: {  }
lcdPower: true
```

expected output:
```
=Promise: {  }
```

myxor, 2021-12-18:
English please.

So far i found this piece of code which looks like the screen is being turned on:

https://github.com/espruino/Espruino/blob/35e46ef6a12ab3dd72e639439ba5f1d409253958/libs/js/banglejs/E_showPrompt_Q3.js#L57

BartS23, 2021-12-19:
Hello,

thanks for the reply and sorry for the wrong language.

Then it is a featrue rather than a bug.

Do you think it is possible to control this behavior via an option?

```
if (!option.keepScreenStatus) {
  Bangle.setLCDPower(1);
}
```

I haven't found a way to check if the backlight is on.

```
if (!Bangle.isLCDOn()) {
  Bangle.setLCDPower(1);
}
```

would work, but does not turn on the backlight.

gfwilliams, 2022-01-04:
> I haven't found a way to check if the backlight is on.

Hmm - good point - I'm not sure there is any way at the moment.

So the issue I think is that `Bangle.setLCDPower(1);` fires an `lcdPower` event even if the LCD is already on? Rather than changing showPrompt/others we should just not fire `lcdPower` if the LCD was already on?

BartS23, 2022-01-10:
Sorry for the late reply.

I think it is an alternative.
Unfortunately I can not serve with a PR, because I have not found the corresponding code.

gfwilliams, 2022-01-11:
Just fixed - looks like the code was intended to only send events on changes, but there was a bug which meant it always fired events

closed 2022-01-11 by gfwilliams