espruino/BangleApps #1139: FEATURE REQ: seperate the LCD and Screen Lock timeouts

hughbarney, 2021-12-18:
I'd quite like to have a short period for the LCD (5 seconds) and say 30 seconds for the screen lock widget.
At present they are the same thing.

To extend battery life I set the LCD timeout to 5 seconds, but when interracting with settings and menus it would be good if the timeout for the lock widget was longer.

RomanistHere, 2021-12-19:
how do you see it?

when you're in settings, LCD becomes off, and you trying to scroll down settings list, what is the behavior you're expecting? Turn LCD on? Don't turn LCD on?

in the first case make timeout 10s - 15s, it won't reduce your battery life noticeable most likely

in the second, just set LCD brightness to 0

gfwilliams, 2022-01-04:
As I understand it the timeouts are actually separate right now: http://www.espruino.com/Reference#l_Bangle_setOptions

So actually it's just that settings doesn't expose it?

DDDanny, 2022-02-04:
I was a bit irritaed too at first. Lock and backlight timings can be different. Of course from my POV it would only make sense to have on a bangle2  backlight of at e.g. 15s and lock after 30s. so backlight<=lock:
When backlight get off you can still touch it to turn light on again.
Nevertheless: The bangle trained the user -> BTN1 is your friend to unlock. Touching helps to prevent lock...

hughbarney, 2022-02-19:
> So actually it's just that settings doesn't expose it?

Yes I guess so.  The link you put in the comment goes to a puck page ?

gfwilliams, 2022-02-21:
> The link you put in the comment goes to a puck page ?

? It goes to the reference for `Bangle.setOptions`

bobrippling, 2023-11-13:
Going to close this for now, but if you're still keen on the functionality, we can re-open it and see if anyone picks it up

closed 2023-11-13 by bobrippling