espruino/BangleApps #1140: Chronos Widget crash

hughbarney, 2021-12-18:
2v11.251 (c) 2021 G.Williams
Uncaught Error: Cannot read property 'value' of undefined
 at line 11 col 78 in chronowid.app.js
timerMenu.hours.value=settingsChronowid.hours;timerMenu.minu...
               ^
in function "predraw" called from line 2 col 79
...20");c.predraw&&c.predraw(g);void 0===b&&c.title&&g.drawStri...
                              ^
in function "draw" called from line 5 col 210
...ax(a,c.selected))}};d.draw();Bangle.setUI("updown",b=>{b?d.m...
                              ^
in function called from line 11 col 955 in chronowid.app.js
...;}},};return E.showMenu(timerMenu);
                                    ^
at line 11 col 955 in chronowid.app.js
...;}},};return E.showMenu(timerMenu);
                                    ^
in function "showMenu" called from line 12 col 10 in chronowid.app.js
showMenu();
         ^
>
{"t":"act","stp":0,"hrm":0}

hughbarney, 2021-12-19:
I had a look at the code but was not able to figure out why this was suddenly happening.
Was on 2.10.251 firmware now on 2.11 and still happens.

elrod16, 2021-12-19:
Can you describe the conditions this is happening under? Like how long a timer, or if it happens when it alerts, etc

hughbarney, 2021-12-28:
Installed the App, selected App from the Launcher, got the above error on startup.

gfwilliams, 2022-01-04:
It looks like the error is because 'predraw' never used to be called, but now it is because `E.showMenu` got changed. I've just removed it and it should be ok now.

commit c06afa11 (2022-01-04): chronowid 0.05: Fix error running app on new firmwares (fix #1140)
closed 2022-01-04 by gfwilliams