espruino/BangleApps #1144: Bike GPS tracker. BangleJS 2.

RomanistHere, 2021-12-19:
I want to create a simple app to track the distance I rode using bike. What would be the best approach to do it?

It will be enough for me to have two buttons: "start" and "finish" - and a place with current distance (passed from start or for the last session if finished)

So what I am basically asking is:

- is it possible? I don't need super accurate results
- how much battery it's going to take for 2h session?
- is the GPS the only way to do it?
- what would be the best time interval to get coords and recalculate the distance?
- if there something similar written already, could you share it with me?
- any other tips would be nice

I would very much like this functionality in the future android app :)

P.S. If you're using existing BangleJS apps with some android apps (exporting and importing data) for the same reasons, I'd be pleased if you shared it with me too. Thanks!

elrod16, 2021-12-19:
I think the GPS Recorder app should tick most of those boxes. It has configurable a gps logging interval, can run in the background, and can plot your recordings on a map (provided you have the right dependencies installed). After a recording you can also look at speed data. If toy need gps plotting and realtime speed data you could use GPS Recorder in the background while using the Speedo app. Be warned though, your first gps fix (no matter what app you use) is going to take a VERY long time, so it'd be best to get that out of the way before a live run.

RomanistHere, 2021-12-21:
[image]

Just wanted to share how I see it

RomanistHere, 2021-12-21:
I think about showing heart rate too, but I read somewhere that it should be measured while resting to have an accurate data. Riding a bike doesn't sound like it, so there is probably no point. However I'm going to test my pulse manually and with BangleJS while riding and compare.

RomanistHere, 2021-12-21:
Actually, on the second thought, it would be nice to save all the data, and probably somehow show sum of every ride somewhere.

Reset functionality needed too in that case

pavelmachek, 2023-05-02:
You may want to take a look at the "run" application. It the functionality you want.

bobrippling, 2023-11-13:
Yes, or the recorder app can do this too, with `bikespeedo` - closing for now

closed 2023-11-13 by bobrippling