espruino/BangleApps #1145: [FEATURE REQ]: Globally scalable fonts

elrod16, 2021-12-19:
Basically what the titles says. It'd be handy to be able to scale any and all (or most at least) texts on screen from a central setting.

gfwilliams, 2022-01-04:
I forget where but there was a discussion about this somewhere else. While in theory it's a good idea, in practice I think most apps wouldn't handle it well (as it's unlikely most authors will test for most sizes). It may make sense to have an option in settings that app authors could choose to honour though.

elrod16, 2022-01-04:
Understandable. Sometimes I forget that the Bangle 2 is just the latest model of an already long existing platform (espruino) and that you have to weigh all the baggage that brings.

Would something like scaling for certain text elements, like the generic prompts the interpreter provides and full screen notifications be something more feasible? There's just some types of info that you want to be able to see with a quick glance, ya know?

Also, I just saw your reply in the corruption thread. Got buried in my email. I'll send that stuff over when I'm back at a computer.

gfwilliams, 2022-01-06:
> like the generic prompts the interpreter provides and full screen notifications be something more feasible?

I think so, yes. Right now I'm thinking maybe we can add something like:

```
g.setSmallFont();
g.setMediumFont();
g.setLargeFont();
```

Which would allow apps to do `g.setFont("Small")`/etc.

Personally I'm not sure you can provide too many useful/good scaling options with the built-in fonts, but having the named fonts means you can just install a 'large font' app that'll provide its own set of fonts for all applications. That should be really neat as it allows you to install (for instance) a themed set of fonts of different sizes.

rigrig, 2022-03-24:
Does it really make sense to globally scale fonts though?
For the messages app, sure: it needs to handle messages which won't fit anyway, so picking a font size makes sense.

But almost all other apps just show some predefined texts, so why shouldn't they just always use the largest font size that will fit?

gfwilliams, 2024-03-15:
Closing this - I think realistically it can be handled by #3109 which is almost there now.

Apps would 100% have to opt in to this anyway as most of them expect fonts to be a fixed size

closed 2024-03-15 by gfwilliams