espruino/BangleApps #1147: Messages chops of first 3 characters off the front of an SMS

hughbarney, 2021-12-19:
Got sent a 6 digit pass code from Amazon.  Only the last 3 digits could be read.

Would take a screenshot but as soon as disconnect gadgetbrudge the bangle says no more messages.
I will try a photo next time.

hughbarney, 2021-12-20:
OK this is a bit more complicated.  There is a general issue with justifying the text into lines.  Here are a couple of examples. Apologies about the poor quality of the photo's.  1) In the first example the bulk of the text is split onto the first line but then the start and end of the first line are clipped off the end.    This example was created using the Android GadgetBridge debug facility to send a message.  2) in the second example I got an email pop up.  The text is rendered only using the inner half of the screen across 4 lines.
[image]
[image]

hughbarney, 2021-12-20:
[image]
[image]

hughbarney, 2021-12-20:
In the 2nd screenshow above the text is:

"Testing and the beanstalk the same time and hear the tragic news"
     What is seen is:
"ting and the beanstalk the same t
    and hear the tragic news"

hughbarney, 2021-12-20:
ok - worked out a fix.  We need to set the font to the font that will be used to display the body of the text before calling g.wrapString();

g.setFont(fontMedium);
lines = g.wrapString(msg.body, g.getWidth()-10);

[image]

gfwilliams, 2022-01-04:
Great - thanks for sorting this! I must not have seen it because I only tested when the title was long so the title font was small

commit 128f2d4e (2022-01-04): messages 0.16: Fix text wrapping so it fits the screen even if title is big (fix #1147)
closed 2022-01-04 by gfwilliams