espruino/BangleApps #1148: Messages App - make fonts configurable through settings

hughbarney, 2021-12-20:
Suggest we adopt the standard names for fonts on the system.
tiny, small, medium, large.

var tinySmall = "6x8";
var fontSmall = g.getFonts().includes("6x15")?"6x15":"6x8:2";
var fontMedium = g.getFonts().includes("12x20")?"12x20":"6x8:3";   // note uplifted to 6x8:3
var fontLarge = g.getFonts().includes("6x15")?"6x15:2":"6x8:4";

This will generally work as below (showing fallback height), 6x8:3 is a better choice for medium as fallback

none                   0                        0   // display nothing
tiny         6x8       8        6x8          8
small      6x15     15      6x8x2      16
medium  12x20   20      6x8x3     24
large       6x15:2  30      6x8:4      32

Next in the messages App the user gets to decide which of these fonts is used in 
msgSource, msgTitle, msgBody

Personally I dont want src/ttitle in the message heading.  I would just want a Medium size Title.

So I would select 
  msgSource:   None
  msgTitle:        Medium
  msgBody:       Medium

gfwilliams, 2022-01-04:
This could be related to https://github.com/espruino/BangleApps/issues/1145 ? 

I feel like maybe we should have something provided globally maybe by `bootloader` that includes font names, then they could be easily changed?

derikb, 2022-02-25:
+1 to this. I just switched over to the new Android/Messages app yesterday and damn are the messages hard to read, the text is so small. While it wasn't the best experience in the gadgetbridge app to not get the full content or even source/name, at least I could see/read it. Now I get a lot of text that I can only read if I really hold it up to my face and squint at.

gfwilliams, 2022-02-25:
Might be worth updating again, there is an option in settings for minimum font size now :)

gfwilliams, 2022-02-25:
Might be worth updating again, there is an option in settings to choose the minimum font size now :)

Actually I think this issue can actually be closed.

closed 2022-02-25 by gfwilliams

derikb, 2022-02-25:
Thanks, that helps. Though I still think for the notification screen those buttons are a LOT of real estate.

gfwilliams, 2022-02-28:
Yes, there are a bunch of forum threads and issues about this and it's something that'll hopefully get looked at soon