espruino/BangleApps #1151: TODO: testframework required

DDDanny, 2021-12-21:
Hey Travis, we need to some
- [ ] lint
- [ ]  chai
- [ ] CI/CD based on xx% fullfillment
Cheers,
Danny

gfwilliams, 2022-01-04:
We already have a linter in CI, along with a bunch of other tests to help make sure things are ok.

What do you mean by chai / xx% ?

DDDanny, 2022-01-06:
Fine with the linter - I'll take a look on it
Regarding xx% 
My guess the lint fullfillment should be 95-100%
For the unit testing a code coverage of e.g. 50%

With Chai I meant chaiJS Test-Framework (https://github.com/chaijs/chai) or Mocca

One example - messages [app.js](https://github.com/espruino/BangleApps/blob/1906dd2839908059fc340555f6f3ae8650570533/apps/messages/app.js#L80) 
```javascript
function getMessageImage(msg) {
  if (msg.img) return atob(msg.img);
  var s = (msg.src||"").toLowerCase();
  if (s=="calendar") return atob("GBiBAAAAAAAAAAAAAA//8B//+BgAGBgAGBgAGB//+B//+B//+B9m2B//+B//+Btm2B//+B//+Btm+B//+B//+A//8AAAAAAAAAAAAA==");
```
[........]
```javascript
  if (msg.id=="back") return getBackImage();
  return getNotificationImage();
}
```

Imagine a new unit-test file like this
naming convention <file_under_test>.test.js - but can also bundle all tests...
-> These files won't be transferred to the clock
-> These files might be executed in the future via webIDE
-> there are no exports and it makes no sense to implement them just for testing, so I guess usage of e.g. [rewire](https://github.com/jhnns/rewire) is "required" ;)

so we make the above function testable via rewire and then
```javascript
var chai = require('chai');
var expect = chai.expect;
var rewire = require("rewire");

var MESSAGES_APP = rewire("app.js");

const msgWithImage = {"img":"test" }
const msgSrcCalender = {"src":"CaLenDaR", "expectedImg": atob("GBiBAAAAAAAAAAAAAA//8B//+BgAGBgAGBgAGB//+B//+B//+B9m2B//+B//+Btm2B//+B//+Btm+B//+B//+A//8AAAAAAAAAAAAA==") }

describe('MessagesApp app.js', () => {

  it('getMessageImage(): should return the img from msg', () => [
    expect(MESSAGES_APP.getMessageImage(msgWithImage)).to.be.equal(msgWithImage.img);
  });

  it('getMessageImage(): should return the expected calendar image', () => [
    expect(MESSAGES_APP.getMessageImage(msgSrcCalender)).to.be.equal(msgWithImage.expectedImg);
  });
});
````

Also the global MESSAGES var can be mocked to test the frontend app behavoiur; inject/reflect the `javascript Bangle.get` states is also possible. With Mocking raise events like `javascript Bangle.On("mag", ...)` I'm unsure but this feature would maybe also interesting for the WEB-IDE for example an emulated compass; so set the heading and see if the app renders correctly.

Is it worthfull to setup a POC for that?
Uhhh that was getting quite long...

gfwilliams, 2022-01-06:
What does everyone else think?

Personally, I doubt more than a handful of people will write test cases for their code, and it could be a pain to run the tests automatically unless it's just running them in a Node.js instance (which doesn't guarantee they work in Espruino). There are a few very niche cases where this might be useful (eg testing some algorithm) but I think in the vast majority of cases (eg testing the image returned as above) this could just hinder other users from making changes.

So...

* Having a system for adding and running tests that run via CI sounds good
* I think it's important we don't force tests to be written as then we just end up with pointless tests that get in the way

I can see real value in being able to spin up an emulator, run the app, push some actions (press button 1, wait 1 second) and compare screenshots automatically, but to me that feels like a pretty huge undertaking.

DDDanny, 2022-02-03:
short update - once i finished work on the timecal clock I'll provide a .test.js sample that automates test of different settings of an class/object based clock/app within the webIDE. 
Basically it runs the testcases by create an instance, execute a function and g.dump() it -> so the "correct" expecgted state can be controlled manually by looking at the webIDE.

gfwilliams, 2022-04-01:
Just some notes on testing:

* In `bin/thumbnailer.js` there's actually some code that can boot the Web IDE's Bangle.js Emulator up within Node.js, run an app and take screenshots.
* There are automatic tests for the `Layout` library in `tests/Layout` that run JS code (using a native Linux build of the emulator), take screenshots and even compare the result.

I guess using the JS emulator (while slower) is probably preferable as there's no guarantee the native one in the repo is kept up to date.

But... IMO there's maybe something to be done there - it could serve two uses - keeping screenshots of apps up to date automatically, but also testing that the apps work as expected.

gfwilliams, 2022-04-01:
Also... we have the sanity checker/lint right now which does a good job, but this could be extended.

* Certain bits of code like `clearInterval()` without arguments should cause an error
* `appname.settings.js` code has to be in a specific form, and we could check for that

gfwilliams, 2024-05-02:
There's code in https://github.com/espruino/BangleApps/blob/master/bin/runapptests.js which will run through tests in an emulator.