espruino/BangleApps #1154: Request: TimeCal: Configure if week starts on Monday or Sunday

m-p-3, 2021-12-21:
I'm used to my calendars starting on Sunday, and would like the ability to configure TimeCal as such. Thanks!

DDDanny, 2022-01-16:
Okay, I'll update the clock for u like this:

0.02: Feature Request #1154 and some findings...
      -> get rendered time from locale
      -> *BATT SAFE* only update once a minute instead of once a second
      -> *RAM optimized* clean code, corrected minute update (timout, no intervall)
      -> locale: weekday name (first two characters) from locale
      -> added settings to render cal view begin day (-1: today, 0:sunday, 1:monday [default])

I personally don't use it:
* Are you able to test the patched clock?
* Is it ok for you that I provide it via my github custom app loader?

Of course I will pre-dev check & test it!

DDDanny, 2022-01-17:
> Of course I will pre-dev check & test it!

I changed a lot, so added multiple customisations of colors and highlights. I still work on some UI(automated) tests.

Preview: 
[image]
the widgets area is here for testing reasons below the clock, but it will dynamically scale it's width. - calendar border is configurable and also sunday color and today's coler

m-p-3, 2022-01-19:
Hi, I checked on https://dddanny.github.io/BangleApps/index.html but TimeCal is still at v0.01 there, would you kindly point me in the right direction? I'll gladly test it on my end :)

DDDanny, 2022-01-19:
I still struggled with my (generic automated UI) testscript. 
The mentioned location is correct. 
I pushed and merged the branch now to my master so you can test in espruino WebIDE and/or with your bangle.

m-p-3, 2022-01-25:
Loaded on my Bangle JS2 running firmware 2v11 Rel

It looks like this
[image]

Also, when going in `Settings > Apps > TimeCal`, it gives me the message

```
TimeCal:
Invalid
settings!

BTN1 to go back
```

DDDanny, 2022-02-01:
Sorry for the late reply !!!
1. I missed the settings app ;) This is working now!
2. Something with the bottom widget recognition is wrong -> I forced +24 now
3. I tested in webIDE and installed successfully - maybe some settings configurations will not work together as expected
4. I kindly ask you to retest

m-p-3, 2022-02-12:
I updated my FW to `2v12`

I can confirm the apps settings now work, and the display is indeed what is expected. Thanks and sorry for the delayed response, I'm still using my Amazfit Bip as my daily driver for now.

commit 63156bcc (2022-02-17): v 0.04 fixes #1154
commit eb9c3ff2 (2022-02-17): Merge pull request #1465 from DDDanny/timecal-clock-0.04

v 0.04 fixes #1154
closed 2022-02-17 by gfwilliams