espruino/BangleApps #1157: Dev Stopwatch app does not work

zloid, 2021-12-21:
@Ronin0000 Hey! Thanks for the Dev Stopwatch app! App really handy and helpful. But at the moment it doesn't work on Bangle.js v1 :(

And how about time counting down endlessly in minutes? For example, with a limit of 24 hours?

Ronin0000, 2021-12-27:
Hi @zloid I am not the creator of the app so I can not help you. You might of got confused that I was the creator because I added a screenshot at was the most recent commit.
[image]

zloid, 2021-12-27:
@Ronin0000 haha :) That's right, sorry :)

zloid, 2021-12-27:
@ukaoskid @fergalwalsh Hi! Maybe you can help with this problem?

ukaoskid, 2021-12-29:
Hey guys, I am gonna take a look to this

DDDanny, 2022-01-11:
_Root Cause_ ...with the bangle2 adoption `setUI` instead of `on watch` the app button handler was brocken.
@ukaoskid seems you are busy and I asked a lot of questions, so I decided to try to close an issue.
And it helps me to get more and more fluent with VS Code and git

PR #1268 (2022-01-12): Devstopwatch app 0.04 - fixes #1157
* *bugfix* stopwatch broken with v0.03 setUI - fixes #1157
* realigned quick n dirty screen positions
* help adjusted to fit bangle1 & bangle2 screen-size with top/lower widgets
* fixed bangle2  foreground color for header: chrono and last lap highlight 3bit color (blue)
* added screenshot for bangle2 and a small README for both
commit eed2d521 (2022-01-12): Merge pull request #1268 from DDDanny/devstopwatch-app-0.04

Devstopwatch app 0.04 - fixes #1157
closed 2022-01-12 by gfwilliams