espruino/BangleApps #1171: widpedom : proposal to use small lato font for large text

hughbarney, 2021-12-24:
I have a version of widpedom that uses the small lato font.
It looks really nice (see below) but not sure if it should go into the codebase as widpedom is one of the standard apps and it needs to work well on all devices.  One possbility would be to have a B1 and B2 version of the code but I dont like that as it creates a maintainance issue for future fixes.

The code change is:
```
$ diff  widpedom.wid.20.js  widpedom.wid.21.js 
117c117,119
<         g.setFont("6x8",2);
---
>         var scale = 1;
>         // Lato from fonts.google.com, Actual height 17 (17 - 1)
>         g.setFontCustom(atob("AAAAABwAAOAAAgAAHAADwAD4AB8AB8AA+AAeAADAAAAOAAP+AH/8B4DwMAGBgAwMAGBgAwOAOA//gD/4AD4AAAAAAAABgAAcAwDAGAwAwP/+B//wAAGAAAwAAGAAAAAAAAIAwHgOA4DwMA+BgOwMDmBg4wOeGA/gwDwGAAAAAAAAAGAHA8A4DwMAGBhAwMMGBjgwOcOA+/gDj4AAAAABgAAcAAHgADsAA5gAOMAHBgBwMAP/+B//wABgAAMAAAAAAAgD4OB/AwOYGBjAwMYGBjBwMe8Bh/AIHwAAAAAAAAAfAAP8AHxwB8GAdgwPMGBxgwMOOAB/gAH4AAAAAAABgAAMAABgAwMAeBgPgMHwBj4AN8AB+AAPAABAAAAAAAMfAH38B/xwMcGBhgwMMGBjgwP+OA+/gDj4AAAAAAAAOAAH4AA/gQMMGBgzwME8BhvAOPgA/4AD8AAEAAAAAAGAwA4OAHBwAAA="), 46, atob("BAgMDAwMDAwMDAwMBQ=="), 21+(scale<<8)+(1<<16));
$ 
$ 
![pedom_68_font](https://user-images.githubusercontent.com/3630481/147351445-9efde13a-ca60-43e0-a9b5-b66899a14789.png)
![pedom_lato_small](https://user-images.githubusercontent.com/3630481/147351458-e14a75a0-56e7-49b3-a47d-f5168524145e.png)

```

elrod16, 2021-12-25:
It does certainly improve readability at a glance or without powering on the LCD backlight.

KTibow, 2021-12-28:
Could you please make a pull request, instead of making an issue with code in it? That way the people who made your code can add your code with one click.

hughbarney, 2021-12-28:
I will be doing just that but I wanted to discuss with @Gordon first as there might be other considerations.

hughbarney, 2022-01-04:
> please don't add this to widpedom.
Ok - understood.  It does look nice but having an uplifted set of quality fonts at global level, would be the way to go.

Just tied Bangle.getHealthStatus("day").steps and I'm not convinced its working properly.

Having done all that testing a while back I have a very good feel for how many steps I have done and today I would have guessed 2000-2500 for today.   5090 is way off.  This is very strange as I thought this was all triggered off the same step counting code ?   How does Bangle.getHealthStatus("day").steps work ?

```
>Bangle.getHealthStatus("day").steps
=5090
>
>WIDGETS["wpedom"].getSteps();
=2447
>
```

hughbarney, 2022-01-04:
```
setInterval(()=>WIDGETS["bata"].draw(), 1000);
Bangle.on('lcdPower', function(on) {
  if (on) WIDGETS["bata"].draw();
});
WIDGETS["bata"]={area:"tr",width:108,draw:function() {
  var sh = Bangle.getHealthStatus("day").steps;
  var sw = WIDGETS["wpedom"].getSteps();
  var steps = '' + sh + ' ' + (sh-sw);
  g.reset();
  g.setColor(g.theme.bg);
  g.fillRect(this.x, this.y, this.x + this.width, this.y + 23); // erase background
  g.setColor(g.theme.fg);
  g.setFont("6x8",2);
  g.setFontAlign(-1, 0);
  g.drawString(steps, this.x, this.y + 12);
}};
```

when I do the final widget I will do the update every 60 seconds.  This will mean that it wont cause additional power draining by updating on every step counted like the Widpedom does.

gfwilliams, 2022-01-05:
Great! I wouldn't have thought updating on the 'step' event would be a massive power draw.

Shame about the `getHealthStatus(day)` issue though - the step count is updated at the same time as other step counts, but it *should* be reset at midnight rather than every 10 minutes. Perhaps there is some issue there though :(

hughbarney, 2022-01-05:
Closed - new issue 1216 raised regarding Bangle.getHealthStatus("day").steps