espruino/BangleApps #1172: "Compact Storage" utility corrupts filesystem

elrod16, 2021-12-24:
I've been able to reproduce it several times now, using the in Settings compact utility and the App Manager compact option.
When executed the wait screen will display, possibly as long as a minute. Then while still on the wait screen the theme will reset to default and afterwards it'll return to the clock screen but it will be a random clock (not the one set) and the app launcher is inaccessible. Upon a forced reboot it stays stuck on the boot screen with a console message about "storage corrupt". Reflashing to default allows the watch to be operated again and to reinstall previous content.

elrod16, 2021-12-25:
After creating this issue again I tried getting a look at the file system to see what exactly was happening, but no luck. The file system was inaccessible from the IDE and trying to load a file manager app into RAM on the watch failed. As soon as it tried loading in code from storage (e.g. widgets) the device froze. The very first time I ran compact storage (which was the first update where it was incorporated into the system settings app) it worked. Every run since (after a flash back to default and all previous apps installed again) it has caused corruption that requires wiping the whole file system to fix. I am running firmware 2v11.251.

m0n5t3r, 2021-12-28:
it worked for me for what it's worth, and made the clock (banglejs 1) responsive again, it used to take 3-4s to react to a press of btn2

hughbarney, 2021-12-28:
I have only used it 2-3 times, has always resulted in some form of corruption that has mean't a complete re-install of all the apps.

elrod16, 2021-12-28:
Lucky! I lose absolutely everything. I got so sick of setting everything up again after each attempt to debug this that I finally wised up and started backing up config files.

And that is interesting that your Bangle 1 has the expected behavior. I am horribly inexperienced with espruino and just flick through the documentation when I have a free minute, so there is a very good chance I'm wrong, but I might have an idea..can anyone point me to where the storage module source is hosted here on github?

I wonder if the module could have a logic error that keeps it on the code branch for bangle 1 instead of correctly branching off for the bangle 2. This is all purely uninformed speculation though.

hughbarney, 2022-01-01:
@elrod - its has happenned a few times for me.  I have always been able to reboot but generally I notice as apps that were installed disappear.  As yet I have not easily reproducable pattern.  We need something like.
1) start with latest firmware 2v11
2) reset to factory defaults
3) install a list of favourite apps
4) do some actions that write to storage (detailed actions needed), this could involve installing, upgrading and deleting apps
5) run the compact process
6) Observe corruption.

I have just tried the above up to step 4 - but have not been able to reproduce.

As it has happenned a few times for me I tend not to use it.
When it has occurred it is usually after a few weeks of installing, upgrading, deleting apps etc etc.

Once we have a recipe that works for more than one person we can give it to Gordon to look into details.

elrod16, 2022-01-01:
When I get around to it I'll try incrementally installing apps. Having the corruption, flash to clean 2v11.251, and just installing my "favorite apps" is enough to ruin the storage if I click "Compact". That's still a lot of variables at this point but for me at least, it doesn't seem dependent on actually writing to storage with any app.

I'm also interested in if the Bangle 1 user above was just lucky or if the Bangle 1 is actually spared from this issue.

hughbarney, 2022-01-02:
I have seen corruption after a compact on Bangle 1 as well. Its one of these issues that's very hard to pin down.
[image]

I'm not sure what the difference is between 2.11 and 2.11.251_cutting edge.  I think this might be some sort of typo.
We were up to 2.10.25x just before 2.11 was finally released.  So its too early to have 2.11.251 as 2.11 has only been released just before christmas and there wont have been any checkins to the firmware so a couple of weeks.

gfwilliams, 2022-01-04:
> I'm not sure what the difference is between 2.11 and 2.11.251_cutting edge. I think this might be some sort of typo.

Yes, they are the same - issue was because I pushed the git change *without* the release tag. Fixed now.

> Having the corruption, flash to clean 2v11.251, and just installing my "favorite apps" is enough to ruin the storage if I click "Compact"

@elrod16 this is great - please could you open the dev console for the app loader, run `SETTINGS.favourites` in the dev console and tell me what it shows? If I can install those apps myself here and reproduce it reliably, I can use that to fix the issue.

elrod16, 2022-01-06:
@gfwilliams hmm, I have tried this several times and it isn't recognizing it is a valid command. I must be doing something wrong. I am connected to my device. Not an emulator or anything. You might need to walk me through this.

gfwilliams, 2022-01-07:
Ahh, that's tricky. I'm not sure how you do the dev console on Android unless you connect it to a PC and do it from there.

Ok, please can you just go to `Library` in BangleApps and click the `Favourites` tag and screenshot that? That should do it I hope

elrod16, 2022-01-07:
Here you go. When counting the system apps I ended up having more favorited than I expected. Hopefully you're still able to do something with this. I'd say a good chunk of my apps list is "standard" stuff most users would have for basic functionality.

gfwilliams, 2022-01-10:
Ok, are you sure this is right? Because I can't even install favourite apps because file manager isn't marked as a bangle.js 2 app

gfwilliams, 2022-01-10:
Just to confirm, I uploaded everything from that list as a favourite (but not file manager because I couldn't), compacted, and it's fine with the latest firmware.

Can you still reproduce? If so maybe I can come up with a way for you to 'image' your Bangle's memory so that I can duplicate it here, run compact, and debug it

SpyGuy0215, 2022-01-14:
I just did this, and I can't fix it! I even reinstalled 2v11 (just installed it 15 mins ago lol) and it's still broken! As of now, still stuck on the wait page :(

elrod16, 2022-01-14:
@SpyGuy0215 Are you still able to connect to the watch via App Loader?

SpyGuy0215, 2022-01-14:
@elrod16 no, had to reinstall firmware

elrod16, 2022-01-14:
So are you stuck on the bangle boot logo? With my watch, I'm typically able to still connect to the App Loader. The only thing that it can do though is format the storage with the tools in the "About" tab. After that I can go about setting it back up.

Also, sometimes after I have fixed and rewritten the storage after this corruption, it'll still hang on the boot logo. The device is actually still running though. Holding down BTN1, as if you wanna navigate back to your clock screen and releasing it early enough you don't force reboot the device, will stop whatever it is hung up on and send you to your clock screen.

gfwilliams, 2022-01-14:
@SpyGuy0215 are you saying it's fixed now you reinstalled firmware? I've never seen that be required, but if you get into a state where it won't boot properly you need to do https://www.espruino.com/Bangle.js2#resetting-without-loading-any-code and then you can connect with the app loader and erase everything

SpyGuy0215, 2022-01-14:
Yeah, everything gone. Stuck on the Bangle boot logo. I couldn't access anything, maybe they were still there but I couldn't access them via the Bangle

SpyGuy0215, 2022-01-15:
Just tried with the compacted in App Manager, "better" result. Bootloader worked, but everything else was gone. When I restarted the Bangle, it did a storage check, said it was corrupt, and erased it. Either way, data's gone.

elrod16, 2022-01-15:
@gfwilliams My device actually succeeded on compacting finally. As far as apps and their possible involvement in this, I had removed "small menus", openstmap, and "Swiper Clock Launch" for reasons other than this issue. Everything else from that list I sent you is still there. Hopefully it wasn't just a fluke.

gfwilliams, 2022-01-27:
Finally this issue is fixed - see https://github.com/espruino/Espruino/issues/2009

closed 2022-01-27 by gfwilliams