espruino/BangleApps #1177: Why does the launcher have a different version for Bangle 1?

KTibow, 2021-12-26:
The default launcher has a version for Bangle 1, and Bangle 2. The version for Bangle 2 seems to be more maintained, and has config options. However, when I edit the app to use the Bangle 2 version, it works just fine for the Bangle 1. Would it be possible to remove the Bangle 1 version and always use the Bangle 2 version?

KTibow, 2021-12-26:
Actually, the up arrow doesn't work. I'll try to figure out why.

peeweek, 2021-12-30:
I can also confirm that issue. At first I thought of a problem of events on my own clock app with 2v11 but it seems related to the app code

KTibow, 2021-12-30:
What issue?

peeweek, 2021-12-30:
Oops sorry, i was referring to :
> Looks like BTN1 now goes back instead of selecting the previous one.

KTibow, 2021-12-30:
Yes, that was an intentional design choice by Bangle:
```js
// pressing button goes back
setWatch(_=>load(), BTN1, {edge:"falling"});
```

peeweek, 2021-12-31:
This seems weird, now we have a menu that can only scroll down? When we use BTN1 on launcher, the previous item is selected, just before going back. Definitely looks like a bug.

KTibow, 2021-12-31:
It definitely is weird, but that's what they chose to do in their code.
https://github.com/espruino/BangleApps/blob/d9d5926416566762e59b953c745f989d23c9b334/apps/launch/app-bangle2.js#L66-L67

KTibow, 2021-12-31:
@peeweek it seems that @t0m1o1 squashed some commits, originally made by Gordon Williams because someone thought it should do that in #971.

t0m1o1, 2021-12-31:
BTN1 on bangle.js 2 is the only button so that's back.
I added settings for fonts to the bangle 2 but I don't have a bangle 1 to test. Does everything work nice other than the wrong button?
Would be nice to unify the code.

KTibow, 2021-12-31:
On the Bangle 1, yes, it functions just like the previous launcher other than having configuration, and using the up button to go back.

peeweek, 2022-01-01:
> On the Bangle 1, yes, it functions just like the previous launcher other than having configuration, and using the up button to go back.
That is precisely what I find strange, on my bangle 1 I just can´t go up anymore because of this change on BTN1 :( . In terms of user experience, I find it way more prevalent to navigate up/down than going back (as I'm already selecting my clock to do so), we could add an extra "return" entry to the launcher menu instead, what do you think?

KTibow, 2022-01-01:
> we could add an extra "return" entry to the launcher menu instead

For me, the clock already displays on the menu as an app.

peeweek, 2022-01-02:
> For me, the clock already displays on the menu as an app.
Seems fair indeed :)
however I can't seem to figure out in the codebase where does the `setWatch(_=>load(), BTN1);` code is handled in the launcher for BangleJS1. Strangely when I run the code in IDE, the button 1 works correctly. Is this back behavior defined more globally (like in boot?)

peeweek, 2022-01-02:
Ok, this feels weirder and weirder.... I am only making assumptions here: I'm feeling that the web updater is not sending the correct js file for the bangle js 1.

I tried to sideload (using the web IDE) the launch.app.js manually by the contents of the app-bangle1.js, and now my menu is working correctly for my banglejs1.

My assumption is that the bangle js 2 version was uploaded to my banglejs1, even though I selected the 1 in the web app

gfwilliams, 2022-01-04:
Ok, so I think there are 2 different issues here? @KTibow there are 2 different versions because the v2 version is designed for touch scroll and the v1 is designed for 3 buttons. I don't think it makes sense to try and use the same version for both.

@peeweek so you think you have the Bangle.js 2 version installed on the Bangle.js 1? Are you using https://banglejs.com/apps/ or your own fork? Maybe check and see if you have launcher settings available from App Settings in the `Settings` app? If so then somehow the Bangle.js 2 version has got installed.

peeweek, 2022-01-08:
> @peeweek so you think you have the Bangle.js 2 version installed on the Bangle.js 1? Are you using https://banglejs.com/apps/ or your own fork? Maybe check and see if you have launcher settings [...]

I've been using both banglejs.com/apps and my own flavor peeweek.github.io/BangleApps/ but I've been selecting correctly the bangle1 on the start screen. I got the problem fixed temporarily by copypasting the bangle1 app.js file into the IDE then uploading it to my watch's storage, and now it works :/ (at least my BTN1 doesn't do both up and return to clock in the menu)

KTibow, 2022-01-08:
> I don't think it makes sense to try and use the same version for both.

AFAIC, you could just change one or two lines of code, and have a universal launcher.
Right now, the Bangle.js 2 version works exactly the same as the Bangle.js 1 version for me, other than BTN1 being used to go back instead of going up.
If a check was added to only use the BTN1 for going to the clock on a Bangle.js 2, then the same launcher could be used for both versions.

KTibow, 2022-01-15:
Going to attempt to unify the two files. Does anyone know how to conditionally run a block of code based on the device version?

t0m1o1, 2022-01-15:
> Going to attempt to unify the two files. Does anyone know how to conditionally run a block of code based on the device version?

From the settings app:
```
 ​const​ ​BANGLEJS2​ ​=​ ​process​.​env​.​HWVERSION​==​2​;
```

True or false for bangle2

PR #1307 (2022-01-18): Unify launcher for bangle.js 2 and 1
## Changes
- Delete separate launcher code for bangle.js 1.
- Slightly modify bangle.js 2 code to work on both devices.
## Issues affected by this
Fixes #1177
(This means bangle.js 1 users will have a way to change the size of the text, and there's less code to maintain.)

- [x] Tested on Bangle.js 1
- [x] Tested on Bangle.js 2 (by @adamschmalhofer)
closed 2022-01-18 by gfwilliams