espruino/BangleApps #1179: Locale for Italian language needs fixing

fparri, 2021-12-27:
I am providing small fixes for the Italian localization. Sorry if I do it here, but I don't know how to do that otherwise :)

1. using : to separate hours from minutes, instead of .
2. in day: there was an extra space character before sabato, showing in certain apps. I removed that.

PS. I suspect the comment in Finnish could be showing the wrong date format, especially in datePattern: { 0: "%A %B %d %Y"  }. Can anyone help with that? :)

Thanks a lot in advance,

  "it_CH": {
    lang: "it_CH",
    decimal_point: ",",
    thousands_sep: ".",
    currency_symbol: "CHF",
    int_curr_symbol: "CHF",
    speed: 'kmh',
    distance: { "0": "m", "1": "km" },
    temperature: '°C',
    ampm: { 0: "", 1: "" },
    **timePattern: { 0: "%HH:%MM.%SS ", 1: "%HH:%MM" }, // 17:00.00  // 17:00**
    datePattern: { 0: "%A %B %d %Y", "1": "%d/%m/%Y" }, // domenica 1. marzo 2020 // 01.03.2020
    abmonth: "gen,feb,mar,apr,mag,giu,lug,ago,set,ott,nov,dic",
    month: "gennaio,febbraio,marzo,aprile,maggio,giugno,luglio,agosto,settembre,ottobre,novembre,dicembre",
    abday: "dom,lun,mar,mer,gio,ven,sab",
    **day: "domenica,lunedì,martedì,mercoledì,giovedì,venerdì,sabato",**
    trans: { yes: "sì", Yes: "Sì", no: "no", No: "No", ok: "ok", on: "on", off: "off" }
  },
  "it_IT": {
    lang: "it_IT",
    decimal_point: ",",
    thousands_sep: ".",
    currency_symbol: "€",
    int_curr_symbol: "EUR",
    speed: 'kmh',
    distance: { "0": "m", "1": "km" },
    temperature: '°C',
    ampm: { 0: "", 1: "" },
    **timePattern: { 0: "%HH:%MM.%SS ", 1: "%HH:%MM" }, // 17:00.00  // 17:00**
    datePattern: { 0: "%A %B %d %Y", "1": "%d/%m/%Y" }, // domenica 1. marzo 2020 // 01.03.2020
    abmonth: "gen,feb,mar,apr,mag,giu,lug,ago,set,ott,nov,dic",
    month: "gennaio,febbraio,marzo,aprile,maggio,giugno,luglio,agosto,settembre,ottobre,novembre,dicembre",
    abday: "dom,lun,mar,mer,gio,ven,sab",
    **day: "domenica,lunedì,martedì,mercoledì,giovedì,venerdì,sabato",**
    trans: { yes: "sì", Yes: "Sì", no: "no", No: "No", ok: "ok", on: "on", off: "off" }
  },

HilmarSt, 2021-12-27:
I'm also new in doing changes with a pull request, but for me it worked like this:
- login to github
- go to the file you want to change, e.g. https://github.com/espruino/BangleApps/blob/master/apps/locale/locales.js
- click on the "pencil" icon left from the "bin" icon to "Fork this project and edit the file"
- edit the file as you had done above and click on "Propose changes"
- fill in some descriptions and press "Create pull request"

If everything worked, you should see your PR in https://github.com/espruino/BangleApps/pulls

fparri, 2021-12-27:
> I'm also new in doing changes with a pull request, but for me it worked like this:

Thanks a lot for teaching me. Done :)))

closed 2021-12-27 by fparri