espruino/BangleApps #1186: Messages disappear after disconnect

m0n5t3r, 2021-12-28:
When getting a message (in my case from the computer via a python script) I'd expect it to stay visible until I dismiss it, or at least until the display timeout passes; right now, a message will  vibrate once regardless of the pattern configured, it gets displayed and goes away when the connection closes, leaving the screen with  "no messages"; if several messages are received in a short time, the last one does vibrate according to the configured pattern, but they disappear just as fast.

Initially I thought it was a matter of the app loading twice or some hidden timeout, but then I tried the same call the script makes (something along the lines of `GB({"t": "notify", "src": "Linux", "title": "Notice", "body": "test"});`) in the IDE, and messages didn't disappear. I then made the script stay connected for a few more seconds, and the messages disappear roughly 1s after the connection closes.

This is ... inconvenient to say the least; is it intentional? is there any workaround for this?

Didgeridoohan, 2022-01-03:
I believe that was introduced as a result of my issue here:
https://github.com/espruino/BangleApps/issues/909

Although since the possibility to delete separate (or all) messages was introduced after that, the "remove messages on disconnect" might no longer be serving much of a purpose (but maybe it should be an option in settings, to cater to different use cases).

m0n5t3r, 2022-01-03:
I thought the problem was read messages popping up again, removing everything is a bit extreme :)

DDDanny, 2022-01-11:
Oh this text got a bit long...

Preamble: 
The #909 is very practical when using gadgetbridge on a android device that is typically "near to the clock". For me it is better to have no more messages as I have them at the phone when not confirmed via bangle - this is the case on a disconnect

1.
> (in my case from the computer via a python script)

So I guess you run a local programm e.g. on a fixed machine and interface with bangle by using adapter "bangle/apps/android" with frontend "bangle/apps/messages".
Of yourse when you leave the BT-range of the fixed device and having it running as a (maybe backend) programm it is not usefull (for you) that the messages are cleared.

2. Btw. thanks for mentioning 
> ..leaving the screen with "no messages"

I created one separate issue  #1267 for that - otherwise this text would be to huge to read

3. So finally - from my POV there are 2 solutions for your custom integration @m0n5t3r 

### Solution 1. android (and ios) adapter apps are getting a setting/sub setting to deactivate that messages are cleared on disconnect.
It is easy to implement but feels a bit of missusing in the context of a external programm (the ios-adapter does the same "clearAll()" on disconnect). Nevertheless it is maybe also usefull for some users with a large home that often forgot their mobile because all they need is on bangle *smile*

### Solution 2. You or someone else create a new generic adapter that provides one message type "notify" and does not clear on disconnect. If you are the owner of the mentioned python programm you can even request a list off all the messages in the python programm and decide which to clear, to be marked read or has to stay unread or flagged as even new on a (re)connect.
This is a bit theoretic because I have absolute no idea what you use your programm and what messages you are exchanging.
If it is e.g. a raspberrypi with smarthome it could be usefull to have dedicated new message types that will be handeled directly by you adapter (e.g. a window open alert or something else)
if your programm is running on a router that also serves telephone calls it can be a use case to give some messages to the messages app
```javascript
... require("messages").pushMessage(event); }
```
and some you will handle with a own widget/app

Storytime ending ;-)

gfwilliams, 2022-01-12:
I think having a setting here makes a lot of sense. We already have a settings menu for Android so adding this should be no big deal. I'll look into adding that.

gfwilliams, 2022-01-12:
sorted now - just change it in Android settings

commit ea44abb7 (2022-01-12): android 0.06: Option to keep messages after a disconnect (default false) (fix #1186)
closed 2022-01-12 by gfwilliams