espruino/BangleApps #1187: Watch wakes up when moved although wakeOnFaceUp and wakeOnTwist are false.

pzo, 2021-12-29:
wakeOnFaceUp and wakeOnTwist are false. 
wakeOnBtn1 and wakeOnTouch are true.
Moving the watch slightly wakes it up, please check enclosed demo. 
Also reproduced after restarting the watch.

https://user-images.githubusercontent.com/3795849/147663170-88536da4-0b00-4c92-9213-f8fe01b58fab.mp4

elrod16, 2021-12-29:
Just to cover the basics, are you on the latest stable firmware (2v11.251) and have the latest settings app updates? And if you are, have you tried going into Settings>Utils>Rewrite Settings?

pzo, 2021-12-30:
Yes, I was on latest stable : 2v11 and all application were up to date using dev site.
Flashed with 2v11.251 and did "rewrite settings" and get the same behaviour.
(Please note, when switching to dfu mode, I still see briefly 2v10.219 on top of the screen even though "about" screen now well displays 2v11.251)

hughbarney, 2022-01-01:
I have observed this, with same config. Normally I only use wake on BTN1 but adding in wakeOnTouch I observed some false wake ups with gerky movements.  It might be the screen sensing a touch when its being vibrated by generate watch movement.

elrod16, 2022-01-01:
Just offering an alternate theory: is it possible that the logic dictating the flow of the program in relation to those condition so has a mistake? Like WakeOnTouch might end up erroneously executing some of the WakeOnTwist or WakeOnFaceUp code? Whether it be an accident assigning variables or an improperly isolated code block.

hughbarney, 2022-01-02:
Its possible.  My theory would be easy to test.  I could just write a small app that counts how many times Bangle.on('touch', fn) gets called and shake your wrist without touching the screen.

pzo, 2022-01-12:
Found the explanation in app/setting/README.MD : 

  * On Bangle.js 2 when locked the touchscreen is turned off to save power. Because of this,
    `Wake on Touch` actually uses the accelerometer, and you need to actually tap the display to wake Bangle.js.

closed 2022-01-12 by pzo