espruino/BangleApps #1191: Calendar app: numbers are barely readable

pasniak, 2021-12-30:
https://github.com/espruino/BangleApps/tree/master/apps/calendar

I would suggest black background instead light blue and increase font by ~30%

[image]

pasniak, 2021-12-30:
Quickly improved version:

[code (8).js.txt](https://github.com/espruino/BangleApps/files/7794442/code.8.js.txt)

[image]

Athemis, 2022-01-05:
That has already been addressed in #1126. However, this version is currently only available from the development version of the app loader (https://espruino.github.io/BangleApps/) and not the stable one (https://banglejs.com/apps).

closed 2022-01-05 by gfwilliams