espruino/BangleApps #1198: Custom QR Code app can’t be removed via App Loader

AppendUser, 2022-01-02:
@halemmerich

Using the App Loader (specifically through the WebBLE app on iOS 15.2), I am able to add the Custom QR Code app to my BangleJS 2 (firmware 2v11) but cannot remove it.

 If I try to use the delete button in the App Loader, it fails with “ Custom QR Code removal failed, ReferenceError: Can't find variable: appListStr” as the output. 

The watch itself will stay on the screen that says “Erasing qrcode” until I hold BTN1 to reset to the clock. Going back to the launcher after this shows the Custom QR Code app and the specific code uploaded still there. 

The only way I’ve been able to remove the app has been to use options which erase all apps in the App Loader. 

[image]

halemmerich, 2022-01-03:
I do not have an iOS device to test that. Using chromium on linux works flawless, just tried removing the app successfully. Does this problem exist for other customized apps? You could try the OpenStreetMap one. Another way to delete the app would be the App Manager.

gfwilliams, 2022-01-04:
Wow, thanks - this is an odd one. Can you delete other apps ok? Do you have a laptop you can try it on? I guess it may be to do with some code we're using not being compatible with Safari (which WebBLE uses under the hood).

elrod16, 2022-01-04:
I have actually ran into this sporadically on Chrome for Android. In my case though, simply going to the "My Apps" tab in the App Loader and doing a manual refresh fixes it. It seems like sometimes the request for the app manifest on the watch gets cut off prematurely.

gfwilliams, 2022-01-05:
Looks like the `appListStr` error is a bug in the error message. I'll get a fix in for that, and then maybe if this re-occurs the error message will be more helpful!

AppendUser, 2022-01-05:
Removing the app through a chromium browser on Windows (Iridium portable) worked as expected, so the problem may be specific to iOS/WebBLE. The App Manager app was also able to remove Custom QR Code without issue.

The OSM app could be uploaded and deleted from WebBLE without problem.

gfwilliams, 2022-01-06:
Odd - just tried on WebBLE here and it deleted the QR Code app fine.

Is this reproducible for you? Do you install the app and then delete right away, or do you install, then disconnect, use the watch, and then it fails when you connect again and try and remove it?