espruino/BangleApps #1204: apps.json merging (custom merge driver)

gfwilliams, 2022-01-04:
Currently if a bunch of PRs are submitted with their json entries at the end of `apps.json` then subsequent ones won't merge automatically and need a small amount of fiddling manually.

It's been discussed in #1020 and other places that we fix this just just not having an apps.json file, but as discussed there, there are a issues with that if we still want to keep using GitHub pages

However it seems that we can define a 'custom merge driver' which will deal with merges to apps.json. For example https://github.com/Praqma/git-merge-driver

We probably only need to have a script that runs the normal git merge command and detects the case where there's a single conflict at the end of the file. In that case it's usually as easy as removing the merge markers and adding a new `}, {` line instead

rigrig, 2022-01-04:
>  it's usually as easy as removing the merge markers and adding a new }, { line instead

I think you also want to parse both conflicting versions as JSON and compare them with the result, to make sure the "dumb" merging didn't mix two apps? (simply parsing them both as JSON and combining would also work, but then you get massive whitespace changes all the time)

reedspool, 2022-01-05:
I only read # 1020 briefly, but I had a thought. If the issue is only permission to do directory listening, then app.json could simply be a list of paths to each app's respective this-app.json file, or even just the app directory name. This could be a JSON array or even a newline delimited plaintext file. Then merging will be trivial.

gfwilliams, 2022-01-05:
There's more info in 1020 I believe and we should probably discuss there but there are workarounds to directory listing on GitHub - personally I think the big issue is having to request 300+ JSON files every time you load the page. I imagine that would really hurt performance.

> This could be a JSON array or even a newline delimited plaintext file. Then merging will be trivial.

Unfortunately the same applies. If two people add a single line to the end of a plaintext file, you'll get a merge error at the moment :(

gfwilliams, 2022-01-06:
>  to make sure the "dumb" merging didn't mix two apps?

I don't think I've really seen any cases where the dumb merge did this? I guess really I'm just thinking we try and fix the one merge issue that happens 95% of the time. The remaining 5% can be merged manually as long as the merge actually fails :)

We could actually merge and set formatting automatically. Formatting always seems to get messed up eventually so doing it as the merge might be an idea anyway

adamschmalhofer, 2022-01-06:
In #1221 I propose a way to have github pages generate the apps.json from per app metadata.json. With that we will avoid all merge conflicts.

rigrig, 2022-01-06:
> I don't think I've really seen any cases where the dumb merge did this?

I wouldn't know, you're the one looking at the conflicts all the time ;-). And I guess the simple replace would fail anyway if things are more messy than expected. So that was just me worrying a bit too much.

> We could actually merge and set formatting automatically. Formatting always seems to get messed up eventually so doing it as the merge might be an idea anyway

While auto formatting would be nice, I definitely think you want to apply it in a separate commit.

Ronin0000, 2022-02-02:
Should this be closed because app.json has been split?

gfwilliams, 2022-02-02:
Thanks - yes!

closed 2022-02-02 by gfwilliams