espruino/BangleApps #1211: [Android Integration + BangleJS2] Gadgetbridge fails to recognize Android Integration

rootabeta, 2022-01-05:
Setup: Samsung Galaxy S21, Android 12. Bangle.js 2, latest bangle2 firmware. 
When using the Gadgetbridge instructions to pair the bangle with the phone, gadgetbridge integration fails. 
Notifications will not load, music information will not display, etc. Debug notifications in the Gadgetbridge app trigger the same behavior. 
Whenever a notification fails to be pushed to the watch, the phone pops up a banner reading "Gadgetbridge
 plugin not installed on Bangle.js"
This behavior does NOT occur when the Gadgetbridge app is installed instead of the Android Integration.

gfwilliams, 2022-01-06:
It sounds like you might be hitting this issue: http://forum.espruino.com/conversations/371523/#comment16337990

That you maybe installed the Gadgetbridge app first, then installed the Android Integration instead, but that's caused some problem.

Please could you disconnect from Gadgetbridge, connect to the Web IDE and paste in this code on the last-hand side?

```
GB({"t":"notify","id":1575479849,"src":"Hangouts","title":"A Name","body":"message contents"})
```

... and let me know what happens? If any messages get shown in the IDE please could you paste them here?

rootabeta, 2022-01-13:
Sorry for the delay, I just saw this.
I ran the code you supplied, and the notification popped up on my watch without issue. Nothing showed in the IDE except the "undefined" return value.

gfwilliams, 2022-01-14:
Thanks - well that all looks ok then... and you're sure it doesn't work on your phone now? Or was this just a one-off error you experienced that isn't reproducable?

rootabeta, 2022-01-14:
It's definitely reproducible, I've tried switching from GB to AI several times with the same results. I reinstalled the Gadgetbridge app on my phone, and now it works with AI, so it may very well be the issue you mentioned. In any case, it seems to be working now.

closed 2022-01-14 by rootabeta