espruino/BangleApps #1215: AppLoader: Search on description

dapgo, 2022-01-05:
The apploader "Search" functions work only with Titles but not with description.
In my opinion  searching on description is necessary, as not all the apps have descriptive name.

A real example of it, is that you can't find the new Gadgetbridge apps because their name is not descriptive.

If searching in both fields is not the expected by everyone, maybe the UI can include a round option to select title/description.

gfwilliams, 2022-01-06:
Sounds like a good idea - it does search tags so I guess all gadgetbridge apps should be tagged as Gadgetbridge as an easy fix though.

Worth noting that the '#' in the URL is used to show up specific apps: https://banglejs.com/apps/#antonclk

We might want to be careful that if we broaden search it doesn't include a bunch of other apps where docs/forum posts/etc have already linked using that URL form. IMO that could be changed somehow to be more explicit - eg something like:

* https://banglejs.com/apps/#antonclk - current behaviour
* https://banglejs.com/apps/?q=antonclk - full search
* https://banglejs.com/apps/?id=antonclk - ID only search

myxor, 2022-02-12:
I am currently implementing this but one question comes into my mind:

Which of the three possible search types (_current/old behavior_, _full search_, _id-only_) should the top search field have?

From my point of view it should be full search (description + id).

gfwilliams, 2022-02-14:
I'd say full search

dapgo, 2022-02-14:
Full would be great but having priority by the location of the string. 
So, items with the string on the title would be sorted over strings on description